introlab/rtabmap #891: Using Lidar with Rtab-map in Winodws c++ based code

triplehoon, 2022-08-26:
I am trying to use both D455 and Lidar in Windows-based system.

Is there any multi-cam example? 

I only find ROS examples but, it is tough for me to work with Cpp-based source code without any examples.

matlabbe, 2022-09-14:
It is recommended to start with ROS for any custom setup including lidar (see tutorials [here](http://wiki.ros.org/rtabmap_ros/Tutorials)). What kind of lidar is it (ring-like or more like a TOF camera)? ROS handles transforms between the sensors (called [tf](http://wiki.ros.org/tf)) and [sensor data synchronization](http://wiki.ros.org/message_filters/ApproximateTime), which rtabmap library doesn't.

If you hard-code your transforms and manually synchronize your data from the two sensors, you could check this c++ example https://github.com/introlab/rtabmap/wiki/Cplusplus-RGBD-Mapping. See also https://github.com/introlab/rtabmap/blob/master/examples/NoEventsExample/main.cpp for an example without threading (easier to understand and modify).

cheers,
Mathieu

closed 2022-09-19 by triplehoon

triplehoon, 2022-09-19:
@matlabbe 
Thanks, I will hard-code and manually synchronize with those examples.