introlab/rtabmap #895: rtabmap-export in CLI does not give full point cloud

Dave-van-der-Meer, 2022-09-05:
Hi,
I created a nice database and I want to extract the point cloud as `.ply` file on the robot through CLI so I can download only the cloud data and not the entire database.
I use the command `rtabmap-export rtabmap.db`. I would expect that I receive the full point cloud but I only get the point cloud of the first frame as it seems.

[image]

When I open the database with `rtabmap-databaseViewer` and then select `File | export map` and keep all the default settings, I get the full point cloud.

[image]

Am I doing something wrong?

Also, when using `rtabmap-export` I receive the following output in the CLI:

```
Loading database "rtabmap.db"...
[ERROR] (2022-09-05 09:10:53.672) VWDictionary.cpp:741::addWordRef() Not found word 1 (dict size=1085)
[ERROR] (2022-09-05 09:10:53.672) VWDictionary.cpp:741::addWordRef() Not found word 1 (dict size=1085)
[ERROR] (2022-09-05 09:10:53.672) VWDictionary.cpp:741::addWordRef() Not found word 2 (dict size=1085)
[ERROR] (2022-09-05 09:10:53.672) VWDictionary.cpp:741::addWordRef() Not found word 2 (dict size=1085)
[ERROR] (2022-09-05 09:10:53.672) VWDictionary.cpp:741::addWordRef() Not found word 3 (dict size=1085)
[ERROR] (2022-09-05 09:10:53.672) VWDictionary.cpp:741::addWordRef() Not found word 3 (dict size=1085)

[...]

[ERROR] (2022-09-05 09:12:55.383) VWDictionary.cpp:741::addWordRef() Not found word 858884 (dict size=1085)
[ERROR] (2022-09-05 09:12:55.383) VWDictionary.cpp:741::addWordRef() Not found word 858885 (dict size=1085)
[ERROR] (2022-09-05 09:12:55.383) VWDictionary.cpp:741::addWordRef() Not found word 858886 (dict size=1085)
[ERROR] (2022-09-05 09:12:55.383) VWDictionary.cpp:741::addWordRef() Not found word 858887 (dict size=1085)
Loading database "rtabmap.db"... done (13.844735s).
Optimizing the map...
Optimizing the map... done (1.668799s, poses=1).
Create and assemble the clouds...
Create and assemble the clouds... done (0.008521s, 12586 points).
Voxel grid filtering of the assembled cloud... (voxel=0.010000, 12586 points)
Voxel grid filtering of the assembled cloud.... done! (0.001121s, 11942 points)
Computing normals of the assembled cloud... (k=20, 11942 points)
Computing normals of the assembled cloud... done! (0.010194s, 11942 points)
Adjust normals to viewpoints of the assembled cloud... (11942 points)
Adjust normals to viewpoints of the assembled cloud... (0.003817s, 11942 points)
Saving ./rtabmap_cloud.ply... (11942 points)
Saving ./rtabmap_cloud.ply... done!
```

How can I export the full point cloud?

Here is the link to the data: https://www.dropbox.com/sh/idoozg2uu6gmvmn/AADgIsvZIhFHDcxd3eT5UYa1a?dl=0

I am using `rtabmap_ros` on a Jetson Xavier NX (Dev Kit) on Ubuntu 20.04  with ROS2 Foxy to create the database.

Please let me know if I can provide more information.

matlabbe, 2022-09-14:
It seems rtabmap was not closed cleanly. You can repair the database with `rtabmap-recovery` tool:
```bash
$ rtabmap-recovery rtabmap.db
[...]
$ rtabmap-export rtabmapleo3.db 

Loading database "rtabmap_leo03.db"...
Loading database "rtabmap_leo03.db"... done (2.647110s).
Optimizing the map...
Optimizing the map... done (0.174467s, poses=356).
Create and assemble the clouds...
Create and assemble the clouds... done (1.643185s, 3644871 points).
Voxel grid filtering of the assembled cloud... (voxel=0.010000, 3644871 points)
Voxel grid filtering of the assembled cloud.... done! (0.316929s, 1765736 points)
Computing normals of the assembled cloud... (k=20, 1765736 points)
Computing normals of the assembled cloud... done! (0.983968s, 1765736 points)
Adjust normals to viewpoints of the assembled cloud... (1765736 points)
Adjust normals to viewpoints of the assembled cloud... (0.967491s, 1765736 points)
Saving ./rtabmap_leo03_cloud.ply... (1765736 points)
Saving ./rtabmap_leo03_cloud.ply... done!

```
[image]

Dave-van-der-Meer, 2022-09-27:
I will try that. I am a bit surprised that the GUI tool can open it. Is it automatically trying o repair it then, when using the GUI? That would explain it I guess. Thanks for the reply.

Dave-van-der-Meer, 2022-09-29:
Indeed, I tried and it works. Thanks!

closed 2022-09-29 by Dave-van-der-Meer

matlabbe, 2022-10-23:
> I will try that. I am a bit surprised that the GUI tool can open it. Is it automatically trying o repair it then, when using the GUI? That would explain it I guess. Thanks for the reply.

The database can be opened with the GUI, but there were missing links and visual dictionary was not saved back into the database.