introlab/rtabmap #897: Building with XCode 14 not working anymore

Godpirate, 2022-09-13:
Hello,
today, after the update of XCode from 13 to 14, i've noticed that is not possible to build the app for an error in the VTK.framework (due to an arm64 error).

I would like to know if it is a problem of the cmake config file, or something regarding the new version of XCode.
This is the error:

error build: In /Users/user/development/rtabmap/app/ios/RTABMapApp/Libraries/lib/vtk.framework/vtk(vtkColorSeries.cxx.o), building for iOS, but linking in object file built for iOS Simulator, file '/Users/user/development/rtabmap/app/ios/RTABMapApp/Libraries/lib/vtk.framework/vtk' for architecture arm64

i've tried a lot of suggested solutions, like excluding the architecture, running XCode with rosetta, rebuilding the libraries, but none of the above worked for me.

matlabbe, 2022-09-13:
You could try to rebuild the dependencies with Xcode14, removing everything in https://github.com/introlab/rtabmap/tree/master/app/ios/RTABMapApp/Libraries (except [install_deps.sh](https://github.com/introlab/rtabmap/blob/master/app/ios/RTABMapApp/Libraries/install_deps.sh)) and redo `install_deps.sh`.

There is a `-DIOS_SIMULATOR_ARCHITECTURES=arm64` on this line : 
https://github.com/introlab/rtabmap/blob/a14b39b9532375af26aa2825e4c6759a0b09e33e/app/ios/RTABMapApp/Libraries/install_deps.sh#L110
Don't remember why it was there, but you may also try `-DIOS_SIMULATOR_ARCHITECTURES=""` so that no simulator version is compiled.

karanbirsingh3391, 2022-09-28:
Please remove the tag -DIOS_SIMULATOR_ARCHITECTURES="" just use below:

cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_FRAMEWORK_INSTALL_PREFIX=$prefix/lib -DBUILD_EXAMPLES=OFF -DBUILD_TESTING=OFF -DVTK_IOS_BUILD=ON -DModule_vtkFiltersModeling=ON ..

Godpirate, 2022-09-29:
Thank you guys, it's working again now.

Closing the issue

closed 2022-09-29 by Godpirate