introlab/rtabmap #900: Debug assertation.

triplehoon, 2022-09-19:
Hello

I am trying to use the example code below.

[https://github.com/introlab/rtabmap/blob/master/examples/RGBDMapping/main.cpp(https://github.com/introlab/rtabmap/blob/master/examples/RGBDMapping/main.cpp)

I successfully modify for my application (using D455 and T265 at the same time with windows application and C++), but It only works on Release mode.

When I build and run the same code with Debug mode, the Debug assertion come out.

```
Debug Assertion Failed!

Program: ~~~~~\Debug\rtabmap_cored.dll
File: C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\include\xtree
Line: 183

Expression: cannot dereference end map/set iterator
```

And when I follow the call stack of the assertion, I find It happened in Rtabamp.cpp

```cpp

std::map<int, Transform> Rtabmap::optimizeGraph(
		int fromId,
		const std::set<int> & ids,
		const std::map<int, Transform> & guessPoses,
		bool lookInDatabase,
		cv::Mat & covariance,
		std::multimap<int, Link> * constraints,
		double * error,
		int * iterationsDone) const
{
...
else
	{
		bool hasLandmarks = edgeConstraints.begin()->first < 0; //<- The assertion breaks here!
		if(poses.size() != guessPoses.size() || hasLandmarks)
		
...
}

```

Any suggestion to using debug mode without any assertions?

matlabbe, 2022-09-19:
This would have to be changed to:
```
bool hasLandmarks = !edgeConstraints.empty() && edgeConstraints.begin()->first < 0;
```
I did it in the commit above. If you find other places, feel free to open a pull request to fix them.

triplehoon, 2022-09-20:
@matlabbe 
Thanks for the fixation.  I will try to open a pull request whenever I find a similar bug.

matlabbe, 2022-10-24:
Re-opening with "enhancement" to test app in Debug config.