introlab/rtabmap #901: OAK-D-Lite won't show up cameras and will freeze the standalone application

Suisse00, 2022-09-22:
Hi,

I tried to use the OAK-D-Lite (TLDR: There is no IMU embedded VS the OAK-D) and it just won't work at all.

When I try to use the "test" or "calibrate" buttons or "start" (_recording_?) there is no camera that will show up and as soon as I stop the action (closing the "test" or "calibrate" window or stop the record) the application just hangs up.

I joined the log (DEBUG level) of a "start"/"stop" record.
[start_stop.log](https://github.com/introlab/rtabmap/files/9621755/start_stop.log)

One potential note that make me into trouble with the DepthAi Demo; I'm using USB2 (there is no USB3 support on the computer running it). The DepthAi Demo work, but UVC, for example, won't.

matlabbe, 2022-10-24:
If the camera doesn't have IMU, the app seems frozen here:
https://github.com/introlab/rtabmap/blob/97a150b1803f9223b2f5ee3fdd7932ca7763df61/corelib/src/camera/CameraDepthAI.cpp#L308-L337

If you can rebuild the standalone, you may change that `while(1)` to a `while(0)` here:
https://github.com/introlab/rtabmap/blob/97a150b1803f9223b2f5ee3fdd7932ca7763df61/corelib/src/camera/CameraDepthAI.cpp#L308

I updated the code in that [commit](https://github.com/introlab/rtabmap/commit/769ace86cedf3ce10ea68bfb7ed85684592930c2) with a new menu option for depthai OAK-D Lite, which disables IMU by default. Also removed the `while(1)` with a check if IMU data is received under 10 ms. 
[image]

Suisse00, 2023-03-08:
If someone happens to read this and have a nigthly build for Windows with the DepthAI support feel free to ping me.
I lost one day trying to compile this without success. (If I start writing what I tried we may be here tomorrow).

Suisse00, 2023-04-01:
OAK-D Lite is working.

For others, trying to use output depth, generating disparity or IMU (this one is on me, I miss clicked) seems to freeze the app.