introlab/rtabmap #906: Nanoflann support

jsseng, 2022-09-27:
I was wondering if there were plans to update the flann library to nanoflann (https://github.com/jlblancoc/nanoflann).  They list some performance and memory optimizations.  When I run rtabmap with a large set of nodes and vocabulary, the load time is longer and I wonder if this update would help.

matlabbe, 2022-10-23:
If there "were" plans, no. However, this could eventually be updated. I added label "enhancement" if someone wants to try it.

matlabbe, 2024-02-20:
Integration idea: There is currently a directory called [rtflann](https://github.com/introlab/rtabmap/tree/master/corelib/src/rtflann) ("**rt**abmap **flann**"), which is a modified version of the flann library with some patches for rtabmap. We could add a new one "nanoflann" containing the [nanoflann header file](https://github.com/jlblancoc/nanoflann/blob/master/include/nanoflann.hpp). Flann is used primarily in [VWDictionary](https://github.com/introlab/rtabmap/blob/master/corelib/src/VWDictionary.cpp) and [RegistrationVis](https://github.com/introlab/rtabmap/blob/master/corelib/src/RegistrationVis.cpp) classes. We either 
1. replace all rtflann related functions by nanoflann, or 
2. make an abstract class for NN and implement common functions using the two approaches. We could add a parameter to select if we want to use rtflann or nanoflann. 

Solution 2 would be the safest transition, as this change can affect the whole loop closure detection performance.

jsseng, 2024-02-20:
I haven't looked at this in a while (and I don't have the time to implement it), but I'm all for adding nanoflann support.  I would be very excited about the save/load functionality and how it would reduce startup times.

matlabbe, 2026-08-16:
PR is merged. I did some refactoring, but nanoflann is mostly useful/faster only for 2D and 3D point clouds. I added some performance tests to compare different flann strategies. It can be ran with `test_flann_index_perf` tool for different scenarios. See description of https://github.com/introlab/rtabmap/pull/1744 to see how to use it for the dictionary (though not recommended).

closed 2026-08-16 by matlabbe