introlab/rtabmap #909: Error(s) building from source with ORB SLAM

rakshith95, 2022-09-30:
Hello, 
I'm trying to build rtabmap with orb_slam2 in a docker, but I get the following errors when compiling:

```
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp: In member function ‘virtual rtabmap::Transform rtabmap::OdometryORBSLAM::computeTransform(rtabmap::SensorData&, const rtabmap::Transform&, rtabmap::OdometryInfo*)’:
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:1002:9: error: ‘class rtabmap::SensorData’ has no member named ‘stereoCameraModel’; did you mean ‘stereoCameraModels’?
 1002 |   (data.stereoCameraModel().isValidForProjection() &&
      |         ^~~~~~~~~~~~~~~~~
      |         stereoCameraModels
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:1003:9: error: ‘class rtabmap::SensorData’ has no member named ‘stereoCameraModel’; did you mean ‘stereoCameraModels’?
 1003 |    data.stereoCameraModel().left().isValidForReprojection() &&
      |         ^~~~~~~~~~~~~~~~~
      |         stereoCameraModels
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:1004:9: error: ‘class rtabmap::SensorData’ has no member named ‘stereoCameraModel’; did you mean ‘stereoCameraModels’?
 1004 |    data.stereoCameraModel().right().isValidForReprojection())))
      |         ^~~~~~~~~~~~~~~~~
      |         stereoCameraModels
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:1021:81: error: ‘class rtabmap::SensorData’ has no member named ‘stereoCameraModel’; did you mean ‘stereoCameraModels’?
 1021 |   CameraModel model = data.cameraModels().size()==1?data.cameraModels()[0]:data.stereoCameraModel().left();
      |                                                                                 ^~~~~~~~~~~~~~~~~
      |                                                                                 stereoCameraModels
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:1022:42: error: ‘class rtabmap::SensorData’ has no member named ‘stereoCameraModel’; did you mean ‘stereoCameraModels’?
 1022 |   if(!orbslam_->init(model, stereo, data.stereoCameraModel().baseline(), imuLocalTransform_))
      |                                          ^~~~~~~~~~~~~~~~~
      |                                          stereoCameraModels
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:1032:25: error: ‘class rtabmap::SensorData’ has no member named ‘stereoCameraModel’; did you mean ‘stereoCameraModels’?
 1032 |   localTransform = data.stereoCameraModel().localTransform();
      |                         ^~~~~~~~~~~~~~~~~
      |                         stereoCameraModels
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:1082:26: error: ‘class rtabmap::SensorData’ has no member named ‘stereoCameraModel’; did you mean ‘stereoCameraModels’?
 1082 |    float baseline = data.stereoCameraModel().baseline();
```

GCC VERSION: `gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0`
CMAKE VERSION: `cmake version 3.16.3`

rakshith95, 2022-09-30:
Changing occurrences of `stereoCameraModel()` to `stereoCameraModels()[0]` seems to solve the build issue, but I don't know if it's the right solution.

matlabbe, 2022-09-30:
Thx for the remember, I somewhat forgot for ORB_SLAM when fixing it for other odometry approaches:
https://github.com/introlab/rtabmap/commit/74c6e9cfecc38f98d50a2441b2b74f534006b194
https://github.com/introlab/rtabmap/commit/675dad5f76b0cd3a35a5bc66cc6dc9ab4f544c39

closed 2022-09-30 by matlabbe

PonyPC, 2023-03-15:
Segmentation Fault after building with ORB SLAM3

PonyPC, 2023-03-15:
```
Thread 35 "rtabmap" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f037e4540 (LWP 21167)]
0x0000007fb7d2047c in rtabmap::StereoCameraModel::baseline() const ()
   from /usr/local/lib/librtabmap_gui.so.0.21
(gdb) bt
#0  0x0000007fb7d2047c in rtabmap::StereoCameraModel::baseline() const ()
    at /usr/local/lib/librtabmap_gui.so.0.21
#1  0x0000007fb7168534 in rtabmap::OdometryORBSLAM::computeTransform(rtabmap::SensorData&, rtabmap::Transform const&, rtabmap::OdometryInfo*) ()
    at /usr/local/lib/librtabmap_core.so.0.21
#2  0x0000007fb7140788 in rtabmap::Odometry::process(rtabmap::SensorData&, rtabmap::Transform const&, rtabmap::OdometryInfo*) ()
    at /usr/local/lib/librtabmap_core.so.0.21
#3  0x0000007fb7144ecc in rtabmap::Odometry::process(rtabmap::SensorData&, rtabmap::OdometryInfo*) () at /usr/local/lib/librtabmap_core.so.0.21
#4  0x0000007fb714e25c in rtabmap::OdometryThread::mainLoop() ()
    at /usr/local/lib/librtabmap_core.so.0.21
#5  0x0000007fb6aa6450 in UThreadC<void>::ThreadMainHandler(UThreadC<void>::Instance*) () at /usr/local/lib/librtabmap_utilite.so.0.21
#6  0x0000007fb59b8088 in start_thread (arg=0x7fffffb54f)
    at pthread_create.c:463
#7  0x0000007fb5ac20cc in thread_start ()
    at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```

PonyPC, 2023-03-15:
use RGBD + Orbslam3 will cause segmentation fault.

matlabbe, 2023-04-02:
For the stereoCameraModels issue, this has been fixed in https://github.com/introlab/rtabmap/pull/991

commit 30bf5895 (2022-09-30): Fixed #909