introlab/rtabmap #910: Parameters to use for RTABMap with ORB SLAM

rakshith95, 2022-09-30:
Hello @matlabbe , 
I've managed to compile OTB SLAM with RTABMap, but running `rtabmap --params | grep orb` only gives me the parameters for the ORB SLAM itself. Is there  a way to ensure orb slam actually runs, and a way to check if it does? 

I've attached the launch file I've been running so far. If you could tell me which parameters I'll have to add/change (or where I can find documentation on it), that would be great.
Thank You!
[stereo_rtabmap_inria.launch.txt](https://github.com/introlab/rtabmap/files/9687756/stereo_rtabmap_inria.launch.txt)

matlabbe, 2022-09-30:
For `rtabmap.launch` include, add this
```xml
<arg name="args" value="-d --Odom/Strategy 5 --OdomORBSLAM/VocPath ~/ORB_SLAM3/Vocabulary/ORBvoc.txt"/>
```
`-d` is to start from fresh database (delete the old one), and adjust the `OdomORBSLAM/VocPath` to where you extracted ORB_SLAM2 or ORB_SLAM3. 

To see if ORB_SLAM is actually used, the odometry view in rtabmapviz will look different.

rakshith95, 2022-10-03:
Thank you @matlabbe! From the log, I can see that the orb vocabulary is now being loaded at least, so I guess it is running ORB SLAM.  I am running it on a docker in a robot, and I did not compile with rtabmapviz.
I have a couple of follow up questions:
1. In #707 you mention the following: 
> We use it like a blackbox (the front-end), which outputs the pose of the current frame that is fed to RTAB-Map's graph management (back-end).  
Is this output pose exposed anywhere, in the form of a topic or sent to TF? 

2. Is ORB SLAM used in both the mapping, and the localization modes?  In the mapping mode, without ORB SLAM, there is a tf update (and pose topic output) only when loop closures are found; does this mean that if run with ORB SLAM, the frequency of tf and the pose updates will be much higher? 

Thank you!

rakshith95, 2022-10-03:
To my first question, I see the topics: 
```
/slam/odom
/slam/odom_info
/slam/odom_info_lite
/slam/odom_last_frame
/slam/odom_local_map
/slam/odom_local_scan_map
/slam/odom_rgbd_image
```
However, the `/slam/odom` topic just seems to publish Identity pose with a very large covariance, and does not update as the camera moves, so I'm not sure what to make of it

rakshith95, 2022-10-06:
Removing the `march-native` flags while building ORB SLAM2, in all the `CMakeLists.txt` files (including third_party repos) seems to have fixed it. Not sure if that directly fixed it, or it was something else though.

closed 2022-10-06 by rakshith95

GVMafeedme, 2023-06-09:
I am trying to load ORB_SLAM3 with RTAB-Map but it does not seem to make a difference, the odometry tab seems to be the same as before, and matches are not being found:

[image]

My launch file has a rtabmap_slam node with these parameters:
```xml
<param name="Odom/Strategy" type="int" value="5"/>
<param name="OdomORBSLAM/VocPath" type="string" value="/root/orb_slam/ORB_SLAM3_NOETIC/Vocabulary/ORBvoc.txt"/>
```
Do I need to add anything else? Are these parameters not correct?

**Note:** I compiled rtabmap and rtabmap_ros from source with ORB_SLAM3 support

matlabbe, 2023-06-11:
With `rtabmap_viz`, set `subscribe_odom_info` to true to get the statistics in odometry panel.