introlab/rtabmap #911: Multi Cam Marker Detection for branch melodic-devel

tugbakara, 2022-10-01:
Hi @matlabbe , I searched multi cam marker detection in rtabmap and saw this [issue](https://github.com/introlab/rtabmap_ros/issues/810), and [here](https://github.com/introlab/rtabmap/commit/4d6bc78e3d8ab8aa66d256937e98bb4e09653031) you committed changes to master only, and while I was looking melodic-devel branch side [code](https://github.com/introlab/rtabmap/blob/melodic-devel/corelib/include/rtabmap/core/MarkerDetector.h) **to be sure** I didn't see any commit, when will they come or just changing codes into melodic-devel branch as master is it worth to try?

matlabbe, 2022-10-01:
You can use `master` branch under melodic. The `melodic-devel` branch is updated only when new melodic binaries are released (`ros-melodic-rtabmap`). To get latest updates, always use `master` branch.

tugbakara, 2022-10-02:
Is it same for rtabmap_ros package?

matlabbe, 2022-10-23:
yes

closed 2022-10-25 by tugbakara