introlab/rtabmap #914: "Global Status Error"--RTABMAP does not work

PhilipAmadasun, 2022-10-08:
I'm trying to use rtabmap with d435 camera with turtlebot2 following these instructions:
```
 $ roslaunch turtlebot_bringup minimal.launch
 $ roslaunch rtabmap_ros demo_turtlebot_mapping.launch
$ roslaunch rtabmap_ros demo_turtlebot_rviz.launch
```
On the rviz display I get Map warning error(message: no map received) and I realize that some. I check and see that rviz is getting image and depth information,so there is scan data being received. I get this error with `roslaunch rtabmap_ros demo_turtlebot_mapping.launch`
```
[ WARN] [1665195508.923794717]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /scan
```
So I assume these two problems are related. Not sure how to fix this issue.

matlabbe, 2022-10-09:
You will have to probably remap the input topics. rtabmap is expecting those topics:
```
/camera/rgb/image_rect_color
/camera/depth_registered/image_raw
/camera/rgb/camera_info
/scan
```
As explained in the warning, you can test if those topics are published with:
```
rostopic hz /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /scan
```
If you need to remap camera topics in this demo, update [those lines](https://github.com/introlab/rtabmap_ros/blob/cf50e931959cda53f88653bc1e884cee6bde890f/launch/demo/demo_turtlebot_mapping.launch#L38-L42). If you don't publish any scan topic, you can remove [that line](https://github.com/introlab/rtabmap_ros/blob/cf50e931959cda53f88653bc1e884cee6bde890f/launch/demo/demo_turtlebot_mapping.launch#L65).

PhilipAmadasun, 2022-10-12:
@matlabbe Id I have to remap, how would I know what to remap to? I'm using realsense d435 camera

matlabbe, 2022-10-23:
```
roslaunch rtabmap_ros demo_turtlebot_mapping.launch \
   rgb_topic:=your_d435_color_image_topic \
   depth_topic:=your_d435_aligned_depth_topic \
   camera_info_topic:=your_d435_color_camera_info_topic
```
You can do `rosnode info your_d435_node` to see which topics are published by the camera. Then set them above.