introlab/rtabmap #918: Export 3D point cloud without GUI via command line

mgow001, 2022-10-26:
Hello, 

I am looking how to export 3D point cloud without using RTAB-map GUI. I know that the standard way of exporting is by clicking through File -> Export 3D clouds -> save. However, in my use-case I can not use the GUI and I am looking for a method that can be executed through a command line along with parameters for exporting point cloud. Is there such a method or a service? 

Also, I know that the alternative option is to save via ROS topic rosrun pcl_ros pointcloud_to_pcd input:=/rtabmap/cloud_map but I'd like to avoid that if possible since it does not allow exporting 3D cloud using custom parameters. 

Thanks!

matlabbe, 2022-11-07:
Check `rtabmap-export` tool, you would find almost all options available in the GUI Export dialog.

mgow001, 2022-11-18:
Hello, just writing to follow-up on this. Is it possible to set `From RGB-D images` parameter using `rtabmap-export` tool? Thanks!

matlabbe, 2022-11-19:
Use `--scan` option

mgow001, 2022-11-28:
That worked, thank you.

closed 2022-11-28 by mgow001

nirmalsnair, 2024-05-22:
Hi @matlabbe,

Is it possible to use a (.ini) config file with `rtabmap-export`?

I'm encountering an issue where exporting a textured (.obj) mesh via `rtabmap-export` results in a file without any faces. However, exporting through the GUI works correctly.

matlabbe, 2024-05-26:
The GUI dialog and `rtabmap-export` don't use the same code, so ini file is not supported with `rtabmap-export` (the options overlap, but I don't guarantee that all same options are in both). Make sure you use `--mesh --texture` options with `rtabmap-export` to create a textured mesh.

nirmalsnair, 2024-05-26:
Thanks @matlabbe, that explains the discrepancies between the GUI and CLI.

Running `rtabmap-export --mesh --texture "new.db"` results in a `.obj` file with no faces and no `.mtl` file.

### GUI vs CLI

[image]

<details>
<summary>GUI log</summary>

```
02:21:44 Clearing memory...
02:21:44 Clearing memory, done!
02:21:44 Connecting to database "C:/Users/centr/Documents/RTAB-Map GUI/new.db"...
02:21:44 Connecting to database "C:/Users/centr/Documents/RTAB-Map GUI/new.db", done!
02:21:44 Loading last nodes to WM...
02:21:44 Loading nodes to WM, done! (8 loaded)
02:21:44 Loading dictionary...
02:21:44 Loading dictionary, done! (1174 words)
02:21:44 Adding word references...
02:21:44 Adding word references, done! (2867)
02:21:47 Downloading the map (global=false ,optimized=true)...
02:21:47 Downloading the map... done.
02:21:47 poses = 8
02:21:47 constraints = 15
02:21:47 Inserting data in the cache (8 signatures downloaded)...
02:21:47 Inserted 8 new signatures.
02:21:47 Inserting data in the cache... done.
02:21:47 Updating the 3D map cloud...
02:21:47 Updated cloud 1 (1/8)
02:21:47 Updated cloud 2 (2/8)
02:21:47 Updated cloud 3 (3/8)
02:21:47 Updated cloud 4 (4/8)
02:21:47 Updated cloud 5 (5/8)
02:21:47 Updated cloud 6 (6/8)
02:21:47 Updated cloud 7 (7/8)
02:21:47 Updated cloud 8 (8/8)
02:21:47 Updating the 3D map cloud... done.
02:21:47 8 locations are updated to/inserted in the cache.

02:22:22 Copied cloud 1 from cache with 3072 points and 2382 indices (1/8).
02:22:22 Copied cloud 2 from cache with 3072 points and 2588 indices (2/8).
02:22:22 Copied cloud 3 from cache with 3072 points and 2597 indices (3/8).
02:22:22 Copied cloud 4 from cache with 3072 points and 2613 indices (4/8).
02:22:22 Copied cloud 5 from cache with 3072 points and 2518 indices (5/8).
02:22:22 Copied cloud 6 from cache with 3072 points and 2618 indices (6/8).
02:22:22 Copied cloud 7 from cache with 3072 points and 2446 indices (7/8).
02:22:22 Copied cloud 8 from cache with 3072 points and 2438 indices (8/8).
02:22:22 Assembling 8 clouds...
02:22:22 Assembled cloud 1, total=2382 (1/8).
02:22:22 Assembled cloud 2, total=4970 (2/8).
02:22:22 Assembled cloud 3, total=7567 (3/8).
02:22:22 Assembled cloud 4, total=10180 (4/8).
02:22:22 Assembled cloud 5, total=12698 (5/8).
02:22:22 Assembled cloud 6, total=15316 (6/8).
02:22:22 Assembled cloud 7, total=17762 (7/8).
02:22:22 Assembled cloud 8, total=20200 (8/8).
02:22:22 Voxelize cloud (20200 points, voxel size = 0.01 m)...
02:22:22 Voxelize cloud (20200 points, voxel size = 0.01 m)...done! (16083 points)
02:22:22 Poisson surface reconstruction...
02:22:22 Poisson depth resolution chosen is 7, map size (m) = 2x1x2
02:22:23 Mesh 0 created with 39812 polygons (1/1).
02:22:23 Transferring color from point cloud to mesh...
02:22:23 Texturing...
02:22:24 Computing visible faces per cam (30719 faces, 8 cams)
02:22:24 Processed camera 1/8: 1669 occluded and 177 spurious polygons out of 19827
02:22:24 Processed camera 2/8: 1784 occluded and 208 spurious polygons out of 19852
02:22:24 Processed camera 3/8: 1894 occluded and 234 spurious polygons out of 18152
02:22:24 Processed camera 4/8: 1928 occluded and 145 spurious polygons out of 17438
[... 50 lines cut ...]
Processed camera 2/8: 2914 occluded and 311 spurious polygons out of 21901
Processed camera 3/8: 3038 occluded and 346 spurious polygons out of 20181
Processed camera 4/8: 2986 occluded and 276 spurious polygons out of 19477
Processed camera 5/8: 3006 occluded and 240 spurious polygons out of 18293
Processed camera 6/8: 2581 occluded and 231 spurious polygons out of 16385
Processed camera 7/8: 2471 occluded and 162 spurious polygons out of 14869
Processed camera 8/8: 2943 occluded and 262 spurious polygons out of 15967
Texturing 34549 polygons...
Textured 8497/10000 of 34549 polygons
Textured 16734/20000 of 34549 polygons
Textured 25671/30000 of 34549 polygons
Texturing... done (0.505868s).
Cleanup mesh...
Cleanup mesh... done (0.053042s).
Merging 8 texture(s)... (1 max textures)
Merging to 1 texture(s)... done (0.859075s).
Saving texture to texture.jpg.
Saved texture.jpg.
Saving obj (20176 vertices) to C:\Users\centr\Documents\RTAB-Map CLI/new_mesh.obj.
```
</details>

The GUI and CLI input/output files are attached below:
[RTAB-Map GUI.zip](https://github.com/introlab/rtabmap/files/15448522/RTAB-Map.GUI.zip)
[RTAB-Map CLI.zip](https://github.com/introlab/rtabmap/files/15448545/RTAB-Map.CLI.zip)

matlabbe, 2024-05-27:
It seems that the CLI is not done yet or crashed while writing the OBJ. The last log message should be:
```
Saved obj to C:\Users\centr\Documents\RTAB-Map CLI/new_mesh.obj!
```
We can see in the OBJ file that is was interrupted writing the last line:
```
vt 0.61655 0.88921
vt 0.61929 0.88837
vt 0.61884 0.89103
vt 0.61673 0.8
```
so the faces were not written yet, and even the MTL is missing (probably written after OBJ). I tried on my computer with `rtabmap-export` and the resulting mesh looks similar to your GUI one:
[image]

On your side, is `rtabmap-export` hanging at the end so you need to do ctrl-c? or just crash silently?

nirmalsnair, 2024-05-27:
Thanks for testing the files.

> On your side, is `rtabmap-export` hanging at the end so you need to do ctrl-c? or just crash silently?

No, `rtabmap-export` does not hang, it just returns to the prompt after
`Saving obj (20176 vertices) to C:\Users\centr\Documents\RTAB-Map CLI/new_mesh.obj.`

nirmalsnair, 2024-05-27:
Your CLI output looks good, so it seems the issue might be with my installation or system specifications.

I tried RTAB-Map on the following two computers:
1. Desktop running Windows 11 with RTX 2070 Super and RTABMap v0.21.4 [CPU-only](https://github.com/introlab/rtabmap/releases/tag/0.21.4) installer.
2. Laptop running Windows 11 with RTX 4090 and RTABMap v0.21.4 (Tried on both [CPU](https://github.com/introlab/rtabmap/releases/download/0.21.4/RTABMap-0.21.4-win64.exe) and [CUDA](https://github.com/introlab/rtabmap/releases/download/0.21.4/RTABMap-0.21.4-win64-cuda117.exe) installers)

[image]
[image]

Could you share the specifications of the system you are using and the version of RTAB-Map you have installed?

matlabbe, 2024-06-03:
> Thanks for testing the files.
> 
> > On your side, is `rtabmap-export` hanging at the end so you need to do ctrl-c? or just crash silently?
> 
> No, `rtabmap-export` does not hang, it just returns to the prompt after `Saving obj (20176 vertices) to C:\Users\centr\Documents\RTAB-Map CLI/new_mesh.obj.`

So it is crashing silently... there could be an issue with PCL 1.14 on Windows (it is crashing inside a PCL function used to write OBJ). The machine specs should not matter. I tested on ubuntu 20.04. 

Can you try version [0.21.0](https://github.com/introlab/rtabmap/releases/tag/0.21.0)? It was built with PCL 1.13.

nirmalsnair, 2024-06-03:
Thanks for your suggestion. I tried both [RTABMap-0.21.0-win64-cuda11.7.exe](https://github.com/introlab/rtabmap/releases/download/0.21.0/RTABMap-0.21.0-win64-cuda11.7.exe) and [RTABMap-0.21.0-win64.exe](https://github.com/introlab/rtabmap/releases/download/0.21.0/RTABMap-0.21.0-win64.exe), but unfortunately, exporting the 3D model does not work from either the CLI **or the GUI**. The face data is still not being fully written, resulting in the same incomplete output as before.

matlabbe, 2024-06-04:
I reproduced your issue on Windows with these two versions. However, the 0.21.0 with old dependencies work: https://github.com/introlab/rtabmap/releases/download/0.21.0/RTABMap-0.21.0-win64-old_deps.zip

I'll open a new issue.

nirmalsnair, 2024-06-04:
Thank you very much! I tried the suggested file, and the mesh export is working fine for both the CLI and GUI.