introlab/rtabmap #919: make -j6 error

Mobilepayload, 2022-11-02:
Hi, I am trying to install rtabmap from `https://github.com/introlab/rtabmap` 
I get this error when I try to run ` make -j6`

`abhi@abhi:~/rtabmap/build$ make -j6
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[  5%] Built target imagesJoiner
[  6%] Built target extractObject
[  6%] Linking CXX shared library ../../bin/librtabmap_core.so
/usr/bin/ld: /opt/ros/noetic/lib/x86_64-linux-gnu/libpointmatcher.a(DataPoints.cpp.o): relocation R_X86_64_PC32 against undefined hidden symbol `_ZTCN5boost10wrapexceptINS_2io12too_few_argsEEE0_NS_16exception_detail10clone_implINS4_19error_info_injectorIS2_EEEE' can not be used when making a shared object
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:1992: bin/librtabmap_core.so.0.20.22] Error 1
make[1]: *** [CMakeFiles/Makefile2:936: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [Makefile:152: all] Error 2`

cmake output:
`abhi@abhi:~/rtabmap/build$ cmake ..
-- MOBILE_BUILD=OFF
CMake Deprecation Warning at CMakeLists.txt:72 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
[... 115 lines cut ...]
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/abhi/rtabmap/build
`

I thought it might be the issue with shared library so I added
 `$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/ros/noetic/lib/x86_64-linux-gnu` this line to my `.bashrc` file according to gyuidelines mentioned in `https://github.com/introlab/rtabmap_ros/tree/master`

Any help is greatly appreciated. Thank You

matlabbe, 2022-11-02:
Can you copy paste the text of the error, it will be easier to search on google for a similar issue. What is your system config? Never saw this problem before.

Mobilepayload, 2022-11-02:
Hi @matlabbe I changed the picture into text 
System-Conifg

Ubuntu 20.04
ROS-Noetic

Let me know if you need anything else, Thank You

Mobilepayload, 2022-11-03:
Hi @matlabbe, I just wanted to know if there is anything you needed from my end?

Thank You

matlabbe, 2022-11-07:
I am not able to reproduce the problem on same system config. Also, the CI is not creating that error either: see latest build https://github.com/introlab/rtabmap/actions/runs/3381479308/jobs/5615429272

CoCube, 2023-01-15:
> I am not able to reproduce the problem on same system config. Also, the CI is not creating that error either: see latest build https://github.com/introlab/rtabmap/actions/runs/3381479308/jobs/5615429272

I have same error by /usr/bin/ld: final link failed: bad value . :( ,i link the lib but it not work

matlabbe, 2023-01-16:
Humm, just tried this minimal `Dockerfile` and it works:
```
FROM ros:noetic-perception

# Install build dependencies
RUN apt-get update && \
    apt-get install -y git software-properties-common ros-noetic-rtabmap-ros && \
    apt-get remove -y ros-noetic-rtabmap && \
    rm -rf /var/lib/apt/lists/

WORKDIR /root/

RUN rm /bin/sh && ln -s /bin/bash /bin/sh

# Copy current source code
RUN git clone https://github.com/introlab/rtabmap

# Build RTAB-Map project
RUN source /ros_entrypoint.sh && \
    cd rtabmap/build && \
    cmake .. && \
    make -j12
```

```
docker build -t test .
```
<details> <summary> Click to show full log </summary>

```
Sending build context to Docker daemon  2.048kB
Step 1/6 : FROM ros:noetic-perception
 ---> 2d806f6ce0fd
Step 2/6 : RUN apt-get update &&     apt-get install -y git software-properties-common ros-noetic-rtabmap-ros &&     apt-get remove -y ros-noetic-rtabmap &&     rm -rf /var/lib/apt/lists/
 ---> Running in bb357b6b61cb
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:2 http://packages.ros.org/ros/ubuntu focal InRelease [4679 B]
Get:3 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:4 http://packages.ros.org/ros/ubuntu focal/main amd64 Packages [760 kB]
Get:5 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [27.7 kB]
Get:6 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [2423 kB]
Get:7 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:8 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
Get:9 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Get:10 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:11 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [979 kB]
Get:12 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [1862 kB]
Get:13 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Get:14 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [2896 kB]
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [31.2 kB]
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1986 kB]
Get:18 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1281 kB]
Get:19 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [55.2 kB]
Get:20 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [28.6 kB]
Fetched 25.8 MB in 6s (4326 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  gir1.2-glib-2.0 gir1.2-packagekitglib-1.0 git-man less libamd2 libappstream4
  libassimp-dev libassimp5 libbtf1 libcamd2 libccolamd2 libcholmod3 libcolamd2
  libcxsparse3 liberror-perl libfreeimage3 libfreenect-dev libfreenect0.5
  libgirepository-1.0-1 libglib2.0-bin libglib2.0-data libgraphblas3 libjxr0
  libklu1 libldl2 liblmdb0 libmetis5 libmongoose2 libogre-1.9-dev
  libogre-1.9.0v5 libpackagekit-glib2-18 libpolkit-agent-1-0
  libpolkit-gobject-1-0 libpyside2-dev libpyside2-py3-5.14 libqt5svg5 libraw19
  librbio2 libshiboken2-dev libshiboken2-py3-5.14 libspqr2 libstemmer0d
  libsuitesparse-dev libsuitesparseconfig5 libtinyxml-dev libumfpack5
[... 1283 lines cut ...]
[ 98%] Built target epipolar_geometry
[ 98%] Built target rtabmap
[ 98%] Built target calibration
[ 98%] Linking CXX executable ../../bin/rtabmap-report
[ 98%] Linking CXX executable ../../bin/rtabmap-matcher
[ 98%] Linking CXX executable ../../bin/rtabmap-rgbd_mapping
[ 98%] Built target report
[ 98%] Built target matcher
[ 98%] Built target rgbd_mapping
[ 99%] Linking CXX executable ../../bin/rtabmap-noEventsExample
[ 99%] Built target noEventsExample
[100%] Linking CXX executable ../../bin/rtabmap-wifi_mapping
[100%] Built target wifi_mapping
Removing intermediate container c70cf6dd00d9
 ---> 83b53f0cd6ab
Successfully built 83b53f0cd6ab
Successfully tagged test:latest
```

</details>

The error [above](https://github.com/introlab/rtabmap/issues/919#issuecomment-1301318126) was related to libpointmatcher, maybe try without it to compare:
```
cmake -DWITH_POINTMATCHER=OFF ..
```

Mk-arc, 2023-02-09:
I have the same issue. Any update on this?

Mk-arc, 2023-02-10:
In my case the problem was related to a self compiled boost version 1.75 I guess. I need boost 1.75 for my own application. During the cmake run RTabMap says its using boost 1.71 which is also installed. But still somehow the install of boost 1.75 is causing this error. If I remove boost 1.75 it somehow works fine.

So I guess this problem is somehow related to libpointmatcher having some problems with boost.

closed 2023-04-15 by matlabbe