introlab/rtabmap #922: Stereo USB Camera with standalone RTabMap

sumitsarkar1, 2022-11-07:
For the standalone RTabMap setup if we select Source as Stereo and Driver as USB can two USB camera be used as a stereo set just like we do for ONLY T265 stereo configuration or this is possible through ROS platform only

rlabs-oss, 2022-11-07:
Standalone will support dual usb cameras, one complication is ensuring both cameras are synchronized and see exactly the same frame at the same time - this normally isn't the case if using two cameras from the same brand via usb e.g webcams.

sumitsarkar1, 2022-11-07:
will ROS platform be helpful for synching the cams and then feed the input to RTabMap?

rlabs-oss, 2022-11-07:
For stereo the fundamental issue to obtain both frames at the same time remains regardless of ROS or standalone.

closed 2022-11-08 by sumitsarkar1