introlab/rtabmap #928: Cannot find Zed in Odometry section for Zed Open Capture

sumitsarkar1, 2022-11-18:
[image]

Did I miss something while installing . Please help.

matlabbe, 2022-11-19:
When building rtabmap, make sure you see at the end of `cmake ..`:
```
  With ZED                  = YES
```

sumitsarkar1, 2022-11-20:
Its NO.....but I was thinking if Zed OpenCapture can make the IMU available on RTabMap...I am on a ONLY CPU platform ...and ZED  SDK requires GPU

matlabbe, 2022-11-27:
Yes, IMU is available on OpenCapture
https://github.com/introlab/rtabmap/blob/860c14e0ecbc1ca79eff783a60d087d742c9dedb/corelib/src/camera/CameraStereoZedOC.cpp#L465-L466
, but not the odometry (which would have been computed by ZED SDK). You should use RTAB-Map's VO instead.

sumitsarkar1, 2022-11-29:
@matlabbe Yes...I am using RTAB-Map's VO on CPU only platforms

closed 2022-11-29 by sumitsarkar1