introlab/rtabmap #929: databaseviewer segmentation fault (core dumped)

tugbakara, 2022-11-18:
Hi @matlabbe !
I install rtabmap latest version when I use rtabmap-reprocess tool to merge two different map I got segmentation fault in terminal:
```
Processed 96/2661 nodes [id=96 map=0 opt_graph=96]... 55ms
[pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
Processed 97/2661 nodes [id=97 map=0 opt_graph=97]... 52ms
Processed 98/2661 nodes [id=98 map=0 opt_graph=98]... 36ms
Processed 99/2661 nodes [id=99 map=0 opt_graph=99]... 38ms
Processed 100/2661 nodes [id=100 map=0 opt_graph=100]... 38ms
Processed 101/2661 nodes [id=101 map=0 opt_graph=101]... 45ms
Processed 102/2661 nodes [id=102 map=0 opt_graph=102]... 43ms
Processed 103/2661 nodes [id=103 map=0 opt_graph=103]... 36ms
Processed 104/2661 nodes [id=104 map=0 opt_graph=104]... 109ms
Processed 105/2661 nodes [id=105 map=0 opt_graph=105]... 53ms
[ERROR] (2022-11-18 12:16:42.286) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 106/2661 nodes [id=106 map=0 opt_graph=106]... 61ms
[ERROR] (2022-11-18 12:16:42.358) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 107/2661 nodes [id=107 map=0 opt_graph=107]... 54ms
[ERROR] (2022-11-18 12:16:42.432) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 108/2661 nodes [id=108 map=0 opt_graph=108]... 56ms
[ERROR] (2022-11-18 12:16:42.519) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 109/2661 nodes [id=109 map=0 opt_graph=109]... 56ms
[ERROR] (2022-11-18 12:16:42.592) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 110/2661 nodes [id=110 map=0 opt_graph=110]... 55ms
[ERROR] (2022-11-18 12:16:42.682) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 111/2661 nodes [id=111 map=0 opt_graph=111]... 51ms
[ERROR] (2022-11-18 12:16:42.757) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 112/2661 nodes [id=112 map=0 opt_graph=112]... 56ms
[ERROR] (2022-11-18 12:16:42.849) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 113/2661 nodes [id=113 map=0 opt_graph=113]... 52ms
[ERROR] (2022-11-18 12:16:42.918) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
[ERROR] (2022-11-18 12:16:42.948) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 114/2661 nodes [id=114 map=0 opt_graph=114]... 78ms Prox on 104 [0]
[ERROR] (2022-11-18 12:16:43.016) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
[ERROR] (2022-11-18 12:16:43.045) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 115/2661 nodes [id=115 map=0 opt_graph=115]... 81ms Prox on 104 [0]
[ERROR] (2022-11-18 12:16:43.123) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
[ERROR] (2022-11-18 12:16:43.147) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 116/2661 nodes [id=116 map=0 opt_graph=116]... 88ms Prox on 106 [0]
[ERROR] (2022-11-18 12:16:43.244) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
[ERROR] (2022-11-18 12:16:43.277) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 117/2661 nodes [id=117 map=0 opt_graph=117]... 98ms Prox on 107 [0]
[ERROR] (2022-11-18 12:16:43.363) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
[ERROR] (2022-11-18 12:16:43.391) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 118/2661 nodes [id=118 map=0 opt_graph=118]... 94ms Prox on 108 [0]
[ERROR] (2022-11-18 12:16:43.478) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 119/2661 nodes [id=119 map=0 opt_graph=119]... 67ms Prox on 109 [0]
[ERROR] (2022-11-18 12:16:43.566) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 120/2661 nodes [id=120 map=0 opt_graph=120]... 65ms Prox on 110 [0]
[ERROR] (2022-11-18 12:16:43.652) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Processed 121/2661 nodes [id=121 map=0 opt_graph=121]... 61ms Prox on 111 [0]
High variance detected, triggering a new map...
[ERROR] (2022-11-18 12:16:43.751) Optimizer.cpp:446::optimizeBA() Optimizer 0 doesn't implement optimizeBA() method.
Segmentation fault (core dumped)
```

and when I open dmesg I got this error:

`rtabmap-reproce[5080]: segfault at 20 ip 00007fc4ebe34fa6 sp 00007ffe1dc931b0 error 4 in librtabmap_core.so.0.20.21[7fc4eba3f000+66b000]
`
I searched this [issue](https://github.com/introlab/rtabmap/issues/187) and there was a bug but this error comes from core lib that's why I suppose there can be installation bug or sth?

tugbakara, 2022-11-18:
**ALSO:**
I mapped in Nvidia and  I tried to reprocess in my pc in nvidia side I mapped with g2o and in my pc I have:
```
rtabmap --version
RTAB-Map:              0.20.21
PCL:                     1.8.1
With VTK:                6.3.0
OpenCV:              4.6.0-dev
With OpenCV xfeatures2d:  true
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With OpenGV:             false
With Madgwick:            true
With PDAL:               false
With TORO:                true
With g2o:                false
With GTSAM:              false
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:           false
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:          true
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:    false
With CCCoreLib:          false
With octomap:            false
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```

matlabbe, 2022-11-19:
I recommend to have at least g2o or gtsam installed. 
```
With g2o:                false
With GTSAM:              false
```

For this error "Optimizer 0 doesn't implement optimizeBA() method.", it means rtabmap is not built with g2o or gtsam, and it is using TORO optimization, which doesn't support BA. It is maybe why the map is not optimized the same between the computers, with TORO it is generally worst.

tugbakara, 2022-11-20:
> Also I had another question but I forgot to mention: When I compare 2D occupancy grid map in RVIZ and databaseViewer tool there are differences they are not huge but important , for example in RVIZ corridors looks nice, no tilting observed but in databaseViewer I saw corridors tilted or not merged enough. Which map do we have to consider actually? I'll try to edit 2D map with databaseViewer's tool (drawing restricted areas) but databaseViewer shows different map compare to RVIZ.

hi @matlabbe  thanks for the answer, but I mentioned another question as I quote above. Wat do you thin about?

matlabbe, 2022-11-27:
I think it is because on RVIZ, rtabmap is launched with some g2o or gtsam support, which correctly optimize the map. When you open on other computer which doesn't have g2o or gtsam, the map is worst as it should use TORO instead. Ideally, have same dependencies on both computers.

tugbakara, 2022-11-28:
Hi @matlabbe , In Nvidia side or in my PC, there exist all optimizers  (in Nvidia only GTSAM is suspicious) when I opened map in databaseViewer **graph view** always shows tilted wherever databaseViewer is. Graph view always looks worse than RVIZ.

matlabbe, 2022-11-30:
Could you share the database?