introlab/rtabmap #935: ios build failed - undefined symbol

diandrakathyusha, 2022-11-29:
Hi, I am trying to build the iOS version of rtabmap to my iPad Pro 2020. I am using a Mac Mini 2014 with MontereyOS. I have installed CMake and OpenCV using homebrew before running the install_deps.sh. I have followed all of the installation instructions, however, I got dozens of the undefined symbols issues in my XCode. I am sure there must be some silly mistake, but I can't figure it out. 

[image]
[image]
[image]

Anyone willing to give any suggestion will be very appreciated!

matlabbe, 2022-11-30:
You don't need "OpenCV using homebrew" (this is for Desktop version), all ios dependencies are installed by `install_deps.sh`. However, I never tried compiling on Intel Mac, so the `install_deps.sh` may not work on Intel platforms.

matlabbe, 2022-12-08:
I just used compiled iOS on an Intel Mac, and it works. Only issue I had is this one: https://github.com/introlab/rtabmap/issues/939

laitifranz, 2023-03-21:
You don't have to run it using the Simulator, otherwise, lots of errors will occur. The same happened to me and I solved it by selecting my physical device in the dropdown devices menu and building the project for it.