introlab/rtabmap #943: Wiki - Security consern over the Windows binaries

Suisse00, 2022-12-13:
Hi,

So I wanted to download a compiled version of the master branch for Windows.
So I went to https://github.com/introlab/rtabmap/wiki/Installation#windows which offers 2 links.

It is odd, both links point to the same file and not in this repo, but whatever it could be a community thing: https://github.com/toxor4/VideoCaptureUtility/releases/download/42/RTABMap-0.20.16-win64.zip

When I unzip it, it is weird, the structure doesn't look likes at all the same as from the github release download from rtabmap.
Then the executable is 800mb with the Adobe Flash logo!?
It asks to run as an admin, ok, could be to install itself? Kinda odd of a nightly build.

My antivirus (Avast) start to go crazy. The first time it is because I crank it up to just be a paranoid antivirus over a ... .bat file trying to be executed in the %TEMP% directory.
I opened the file with a text editor, and hum, I see red flags pretty much everywhere.

It uses a different shell variable after each damn character. If you remove them, you end up with some nice commands.
Two powershell commands with a base64 command (instead of clear text). One downloading  w ww.rv mis.com/vend or/7za.exe & w ww.rv mis.com/vend or/re d.zip (remove space from them)

Then it unzip the file, that contains a password (another red flag) and then it try to execute an executable from the zip.

matlabbe, 2022-12-14:
That is bad. There many users changing the links over time... :( 

It seems those users/bots don't exist anymore:
* https://github.com/zekarysalad
* https://github.com/vipclick360maps
* https://github.com/PitVital
* https://github.com/betafIight
* https://github.com/vendiq

I removed "everyone" permissions to edit the wiki pages... trying to repair all links...

matlabbe, 2022-12-14:
I am so sorry for all the troubles. The links should all be repaired by now.

closed 2022-12-14 by matlabbe