introlab/rtabmap #948: Implement multi cameras in standalone version

fjlxb, 2022-12-20:
Hello, 

I want to use multi camera in standalone, for some reasons I can not use ROS, yes I have read your answer in #907 , but still do not know how to make it in standalone version. (My machine supports RGBD/Stereo)

Thank you~

matlabbe, 2022-12-21:
With RTAB-Map main App, it will be difficult. You would have to adapt one of these [examples](https://github.com/introlab/rtabmap/tree/master/examples). Assuming you have two identical ZED cameras (it is important that all cameras have same resolution to be compatible with multi-stereo or multi-rgbd camera approach of rtabmap), you may create two [CameraStereoZed](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/camera/CameraStereoZed.h) cameras, initialized with right serial ID (and right local transform: base link -> to camera link). You will need to capture **at the same time** from both cameras 2 SensorData (using [takeImage()](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/Camera.h#L54) function). Once you have 2 SensorData (one for each camera), you have to merge them into a single SensorData. For the new SensorData, just add the two camera models (or the two stereo camera models) with correspondinf function. for the left/rgb and right/depth image, you have to stitch them into a single left/rgb image and right/depth image:
```
Combined Left (cv::Mat(height, widthx2)):
-----------------------------
|             |             |
|CameraA: left|CameraB: left|
|             |             |
-----------------------------

Combined Right (cv::Mat(height, widthx2)):
-------------------------------
|              |              |
|CameraA: right|CameraB: right|
|              |              |
-------------------------------
```
or for RGB-D mode:
```
Combined RGB (cv::Mat(height, widthx2)):
-----------------------------
|             |             |
|CameraA: rgb |CameraB: rgb |
|             |             |
-----------------------------

Combined Depth (cv::Mat(height, widthx2)):
-------------------------------
|              |              |
|CameraA: depth|CameraB: depth|
|              |              |
-------------------------------

```
Then set those combined images to new SensorData. Publish that sensor data to Odometry object.

fjlxb, 2022-12-28:
Oh, thank you, I will have a try

fjlxb, 2022-12-28:
> With RTAB-Map main App, it will be difficult. You would have to adapt one of these [examples](https://github.com/introlab/rtabmap/tree/master/examples). Assuming you have two identical ZED cameras [...]

Oh, thank you, I will have a try

commit caac2ee0 (2023-05-06): Fixed #948
closed 2023-01-03 by fjlxb