introlab/rtabmap #950: Failed to build rtabmap with VTK-9.2.2 and PCL-1.13.0 Ubuntu22.04

zhangzscn, 2022-12-30:
have Make sure VTK 9.2.2 is built with Qt

make -j32
Consolidate compiler generated dependencies of target rtabmap_utilite
[  4%] Built target rtabmap_utilite
Consolidate compiler generated dependencies of target res_tool
Consolidate compiler generated dependencies of target imagesJoiner
Consolidate compiler generated dependencies of target extractObject
[  5%] Built target res_tool
[  6%] Built target imagesJoiner
[  6%] Linking CXX executable ../../bin/rtabmap-extractObject
/usr/bin/ld: 找不到 -lvtkGUISupportQt: 没有那个文件或目录
Consolidate compiler generated dependencies of target rtabmap_core
[  6%] Linking CXX shared library ../../bin/librtabmap_core.so
collect2: error: ld returned 1 exit status
make[2]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/build.make:166：bin/rtabmap-extractObject] 错误 1
make[1]: *** [CMakeFiles/Makefile2:960：tools/ExtractObject/CMakeFiles/extractObject.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
/usr/bin/ld: 找不到 -lvtkGUISupportQt: 没有那个文件或目录
collect2: error: ld returned 1 exit status
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:1953：bin/librtabmap_core.so.0.20.22] 错误 1
make[1]: *** [CMakeFiles/Makefile2:826：corelib/src/CMakeFiles/rtabmap_core.dir/all] 错误 2
make: *** [Makefile:156：all] 错误 2

closed 2023-01-03 by zhangzscn

tugbakara, 2023-01-06:
Actually I am trying to install PCl , OpenCV with CUDA support when rtabmap_ros comes into play, I couldn't make catkin_make that's why I suspcious about PCl (includes changed VTK, Eigen etc...) versions cause this issue, how did you solve it , or could you tell me which lib versions (PCL, VTK etc...) can be compatible wtih rtabmap, rtabmap_ros side? 
Also @matlabbe  , could you look at that issue?

matlabbe, 2023-01-09:
For Ubuntu 22.04, normally rtabmap is able to build, see https://github.com/introlab/rtabmap/actions/runs/3869686505/jobs/6595977316

Default versions on 22.04 are PCL 1.12.1,  VTK 9.1, Qt 5.15.3 and OpenCV 4.5.4. For custom versions, I cannot really test all of them or be up to date (will eventually try them when I need it or when new ubuntu versions are released).