introlab/rtabmap #953: Fuse with GPS data in standalone version

fjlxb, 2023-01-06:
Hello, @matlabbe 

I know rtabmap support fusing GPS with /gps topic in ROS version,

however, in my situation, the mechine can not run ROS. So the question is, how can I fuse GPS data in RGBD data? Is there any exmaples to follow?

Thanks.

matlabbe, 2023-01-09:
Do you have a c++ API to read your GPS? for the standalone, a hacky way could be to read the GPS and set it in [SensorData](https://github.com/introlab/rtabmap/blob/98bf3fb184f89e8061b3d985f96a6c4f62ed6749/corelib/include/rtabmap/core/SensorData.h#L286) published by odometry thread:
https://github.com/introlab/rtabmap/blob/98bf3fb184f89e8061b3d985f96a6c4f62ed6749/corelib/src/OdometryThread.cpp#L126
or set it in rtabmap thread when receiving new data:
https://github.com/introlab/rtabmap/blob/98bf3fb184f89e8061b3d985f96a6c4f62ed6749/corelib/src/RtabmapThread.cpp#L473

fjlxb, 2023-01-10:
OH, I have a try~thank you~

closed 2023-01-16 by matlabbe