introlab/rtabmap #954: How to change the Mesh color to green instead of default red/maroon?

Kushjadiaa, 2023-01-12:
I want to change the scanned area color to Green/Blue instead of the default Red/Maroon color. How can I change that color in iOS using code with swift?

matlabbe, 2023-01-16:
Search for `uRedUnknown` in https://github.com/introlab/rtabmap/blob/master/app/android/jni/background_renderer.cc. It is quite hard coded in the shader.

Kushjadiaa, 2023-01-16:
Hey @matlabbe, I found the "uRedUnknown" variable. But I think it is for **Android** users. I need a solution in the iOS. 

How can I change that color in iOS Lidar devices? Can you change the red color to green and share the code example it would be very helpful? 

Thanks!

matlabbe, 2023-01-17:
That code is the one used also for iOS. I didn't have time to move the common iOS/android code into a "common" subfolder outside android and ios subfolders.

Kushjadiaa, 2023-01-17:
@matlabbe I've changed **uRedUnknown** to **uBlueUnknown** and **uGreenUnknown** but unfortunately it is showing the red color mesh. Can you please suggest a way to change the color of the mesh? 

**Image with Red color mesh**
[image]

**Code snippet**
[image]

matlabbe, 2023-01-17:
You have to change the gl_FragColor. The boolean `uRedUnknown` is not the color, it is use din the gl_FragColor formulat to set Blue and Green channels to 0. If you want Blue instead of Red:
```
gl_FragColor = vec4(uBlueUnknown?0.0:grey, grey, uBlueUnknown?0.0:grey, 0.5);
```