introlab/rtabmap #956: iOS | Want to show scanned area with Red Mesh/ Red Dot

Kushjadiaa, 2023-01-17:
I want to show the Red Mesh to the scanned area in iOS iPhone 13 Pro, like whatever data scanned needs to be added as a red dot, and not scanned data should be kept as it is whatever it is really.  

How can I achieve this with the code in iOS swift?

matlabbe, 2023-01-17:
Not sure exactly what you want to do, but all 3D rendering code is done in c++/openglES under android/jni directory (all cpp code here is included in iOS):
https://github.com/introlab/rtabmap/tree/master/app/android/jni

Kushjadiaa, 2023-01-18:
My requirement is to cover all the area which is scanned with the Red mesh, and whichever area is not scanned should appear as it is seen (without the Red Mesh). I am attaching one image in which I have performed scanning. In that Red mesh is shown on the portion which is not scanned. So basically, I want to do the vice-versa of that.

How can I achieve this using iOS Swift?
Please guide me on this. 
[image]

rlabs-oss, 2023-01-18:
Kushjadiaa Isn't rtab-map ios already doing that? e.g colour scan for regions already covered (no dots) - with red dots showing incomplete scan regions.

Kushjadiaa, 2023-01-19:
@rlabs-oss  @matlabbe Yes, rtab-map is doing that, but I want to do the exact reverse of that. Red dots should appear on the scanned region and the region not scanned should be without red dots.