introlab/rtabmap #958: Low frequency using ros2 rtabmap

takahiko-hasegawa-hdjp, 2023-01-22:
I'm currently using (ros2)foxy version of rtabmap with the default rtabmap.launch.py to do the localization but seems like it's having some issue with the frequency. 

I was expecting about 10Hz (since I managed to get 10Hz with ROS1) but looks like the average frequency is less than 1Hz ->

 average rate: 0.793
        min: 0.686s max: 3.370s std dev: 1.00981s window: 13

I followed the instruction on http://wiki.ros.org/rtabmap_ros/Tutorials/Advanced%20Parameter%20Tuning to increase the speed but ha s no effect. 

Is there anything that I could look at to increase the speed?

Thanks in advance.

matlabbe, 2023-01-23:
Which topic are you monitoring frequency? Make sure to build the package in Release mode.

takahiko-hasegawa-hdjp, 2023-01-23:
im looking at rtabmap/mapGraph 
and for the build process, I just used colcon build for rtabmqap and rtabmap-ros

matlabbe, 2023-01-23:
Well, by default `rtabmap/mapGraph` is published at 1 Hz (`Rtabmap/DetectionRate`). To build in release mode, make sure [to use](https://github.com/introlab/rtabmap_ros/tree/ros2#installation):
```
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
```

takahiko-hasegawa-hdjp, 2023-01-23:
Just out of curiosity, I changed the Rtabmap/DetectionRate to 10 (not in release just for testing), but no effect. I suppose release build is a MUST to get the frequency up?

matlabbe, 2023-01-23:
Building in release makes the code run faster, but on foxy, you may be already in release by default. Other causes can be that input topics are poorly synchronized, causing delays and then low output frequency. In the logs, what is the RTAB-Map processing time?

takahiko-hasegawa-hdjp, 2023-01-23:
Hi there, built in release mode but still the same thing. 
The RTAB-MAP processing is roughly around 0.75s 
the camera frame is coming in at 10Hz

matlabbe, 2023-01-23:
I don’t what is your setup (computer, what kind of camera, is database starting empty), but 0.75 is too low… unless you try to process 4K images. If you can record a rosbag of your input topics, that would be useful to debug more… if it is ros2 version or something else.

takahiko-hasegawa-hdjp, 2023-01-24:
the input image is (1280x720) for both depth and  rgb. the database is always startyong empty.

matlabbe, 2023-01-28:
Ideally...

> If you can record a rosbag of your input topics, that would be useful to debug more