introlab/rtabmap #959: Unable to build with c++17

Neumann-A, 2023-01-23:
rtabmap seems to use a bunch of deprecated functions blocking the build using Qt6

D:\buildtrees\rtabmap\src\v0.20.16-df5c73836d.clean\corelib\src\rtflann/util/heap.h(118): error C2039: 'binary_function': is not a member of 'std'

D:\buildtrees\rtabmap\src\v0.20.16-df5c73836d.clean\corelib\src\rtflann/util/random.h(120): error C2039: 'random_shuffle': is not a member of 'std'

D:\buildtrees\rtabmap\src\v0.20.16-df5c73836d.clean\corelib\src\rtflann/algorithms/dist.h(522): error C2065: 'pop_t': undeclared identifier

matlabbe, 2023-01-23:
Try master branch, c++17 should work

Neumann-A, 2023-01-23:
> Try master branch, c++17 should work

No it won't. Your vendored flann is outdated and still uses std::binary_function and random_schuffle. 
https://github.com/introlab/rtabmap/blob/ff61266430017eb4924605b832cd688c8739af18/corelib/src/rtflann/util/heap.h#L118
https://github.com/introlab/rtabmap/blob/ff61266430017eb4924605b832cd688c8739af18/corelib/src/rtflann/util/random.h#L120

just to point out two occurrences in the code.

matlabbe, 2023-01-23:
You are right, for some reason I thought I already built it on c++17, but it was c++14
https://github.com/introlab/rtabmap/blob/ff61266430017eb4924605b832cd688c8739af18/CMakeLists.txt#L746-L755

I'll check to update flann, hopefully it won't break older machine configurations.

matlabbe, 2023-02-05:
Well, on ubuntu 20.04, setting explicitly to c++17 doesn't cause any building issues:
```
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++17
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DDISABLE_OPENNI2;-DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0
--   PCL_VERSION = 1.10.0
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4.2.0 xfeatures2d = NO, nonfree = NO (License: BSD)
--   With Qt 5.12.8            = YES (License: Open Source or Commercial)
--   With VTK 7.1              = YES (License: BSD)
```
I'll try with in a docker 22.04 + Qt6...

Neumann-A, 2023-02-05:
If it compiles with the stuff mentioned in https://github.com/introlab/rtabmap/issues/959#issuecomment-1400698501 and c++17 then your c++ library is non conforming. But you probably want to check the build logs instead of the configure log if c++17 is really applied and cmake didn't add c++14 add the end.

matlabbe, 2023-02-06:
I've been updating to Qt6 on this branch https://github.com/introlab/rtabmap/tree/qt6 with latest VTK/PCL and OpenCV, using c++17, and I still don't see the errors (this is a docker on Ubuntu 22.04)...
```
[  5%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/VWDictionary.cpp.o
cd /root/rtabmap/build/corelib/src && /usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DDISABLE_OPENNI2 -DDISABLE_PCAP -DDISABLE_PNG -DG2O_USE_VENDORED_CERES -DPCL_OMP -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DQT_OPENGLWIDGETS_LIB -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -Dkiss_fft_scalar=double -Drtabmap_core_EXPORTS -I/root/rtabmap/build/corelib/src/../include -I/root/rtabmap/utilite/include -I/root/rtabmap/corelib/src/../include -I/root/rtabmap/corelib/src -I/root/rtabmap/build/corelib/src -I/usr/include/openni2 -I/usr/include/dc1394 -I/usr/include/hidapi -isystem /usr/local/include/opencv4 -isystem /usr/local/include/pcl-1.13 -isystem /usr/include/eigen3 -isystem /usr/include/ni -isystem /opt/ros/humble/include -isystem /usr/local/include/vtk-9.2 -isystem /usr/local/include/vtk-9.2/vtknlohmannjson/include -isystem /usr/local/include/vtk-9.2/vtkfreetype/include -isystem /usr/include/x86_64-linux-gnu/qt6/QtOpenGL -isystem /usr/include/x86_64-linux-gnu/qt6 -isystem /usr/include/x86_64-linux-gnu/qt6/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt6/mkspecs/linux-g++ -isystem /usr/include/x86_64-linux-gnu/qt6/QtGui -isystem /usr/include/x86_64-linux-gnu/qt6/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt6/QtOpenGLWidgets -isystem /opt/ros/humble/include/g2o/solvers/pcg -isystem /opt/ros/humble/include/g2o/solvers/csparse -isystem /usr/include/suitesparse -isystem /opt/ros/humble/include/g2o/solvers/cholmod -fmessage-length=0 -fopenmp -std=c++17 -O3 -DNDEBUG -fPIC   -Wall -Wno-unknown-pragmas -msse4.2 -mfpmath=sse -march=native -mavx2 -fPIC -fopenmp -MD -MT corelib/src/CMakeFiles/rtabmap_core.dir/VWDictionary.cpp.o -MF CMakeFiles/rtabmap_core.dir/VWDictionary.cpp.o.d -o CMakeFiles/rtabmap_core.dir/VWDictionary.cpp.o -c /root/rtabmap/corelib/src/VWDictionary.cpp
[  5%] Linking CXX shared library ../../bin/librtabmap_core.so
cd /root/rtabmap/build/corelib/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/rtabmap_core.dir/link.txt --verbose=1
/usr/bin/c++ -fPIC  -fmessage-length=0 -fopenmp -std=c++17 -O3 -DNDEBUG -shared -Wl,-soname,librtabmap_core.so.0.20 -o ../../bin/librtabmap_core.so.0.20.23
```

Maybe it is related to compiler on windows? I'll continue to keep on eye on this, and fix it when I will get the error.

matlabbe, 2023-03-17:
@Neumann-A I integrated your changes on latest master. The new 0.21.0 version is also now using cmake targets. I tried vcpkg build and cleanup the patches to build everything. See my pull request here: https://github.com/microsoft/vcpkg/pull/30254. If you can comment on it it would appreciated.

With VCPKG, the only issue I have right know is that Qt6 plugins are not copied in `tools/rtabmap`, so if you launch any rtabmap's app/tool using qt, you will have a qt error that `windows` plugin is not found. A workaround is to set `QT_PLUGIN_PATH` as an environment variable (for me it is set to `C:\vcpkg\installed\x64-windows-release\Qt6\plugins`)

I tried https://github.com/microsoft/vcpkg/issues/20101#issuecomment-1077747306 in rtabmap's port file but I got cmake error during the build. I don't know where to put this.

Neumann-A, 2023-03-17:
Have you tried to use https://www.qt.io/blog/cmake-deployment-api for deployment?

Another solution might be to just copy the `qt.conf` from `tools/Qt6/bin`

matlabbe, 2023-03-18:
Thank you for the reviewing the [PR](https://github.com/microsoft/vcpkg/pull/30254), I am currently making the changes. For qt deployment, I missed that webpage, I tried the first time [this](https://doc-snapshots.qt.io/qt6-6.5/qt-deploy-runtime-dependencies.html), but saw after testing it was only for Qt 6.5, while vcpkg uses Qt 6.4. Then I tried [this](https://doc-snapshots.qt.io/qt6-6.4/qt-deploy-runtime-dependencies.html), which ultimately worked in my bundle build https://github.com/introlab/rtabmap/commit/46edd15121b5efc607f3141e61e54171c48b791b, but it didn't work when I used the same cmake code in the `portfile.cmake`. The approach in your [link ](
https://doc.qt.io/qt-6/qt-generate-deploy-app-script.html) looks cleaner, I'll try that for vcpkg and also refactor the previous approach I used for the bundle (as I don't like the hard-coded Release).

matlabbe, 2023-03-18:
Closing as it now works with C++17.

closed 2023-03-18 by matlabbe

mattiasmar, 2023-06-09:
@matlabbe:  What Dockerfile did you use to test this?

matlabbe, 2023-06-11:
To test latest dependencies, see this docker file: https://github.com/introlab/rtabmap/blob/master/docker/latest_deps/Dockerfile. Note that this docker file is not built in CI (would be too long) and as it is using always latest dependencies, it may fail at some point if dependencies have regressions (if so, open another issue). To build it:
```
cd rtabmap
docker build -t rtabmap_latest -f docker/latest_deps/Dockerfile .
```