introlab/rtabmap #963: iOS: First Person view disabled after loading an old map

matlabbe, 2023-01-25:
First Person view disabled after loading an old map. However, the First Person view is enabled just after we Assemble the map (or save and optimize just after a new scan). As a workaround, re-assemble the map after loading it, the First Person view would be then enabled for VR visualization.

May also affect tango version too (yet to be confirmed).

matlabbe, 2023-01-29:
Tested on iPad 2020 and First Person view is enabled after loading an old map. Have to try again with newer iPad/iPhone.

For Tango, we should set **Mode->Localization**, then do **Resume** to enter VR mode while keeping the optimized mesh. I don't think the new approach from iOS has ever been back ported to android version.

xxxEWxxx, 2023-01-30:
I can confirm that First Person view is disabled after loading an old map on iPhone 13 Pro Max (iOS 16.0).

Kushjadiaa, 2023-01-30:
@xxxEWxxx I am facing the same issue in iPhone 13 Pro (iOS 16.3)

xxxEWxxx, 2023-01-30:
Seems like First Person View requires an optimized mesh: 
https://github.com/introlab/rtabmap/blob/05676d1577228f0981d32ab7587ae1e0b82ad500/app/android/jni/RTABMapApp.cpp#L413
If a database is safed only and not optimized, the first person view (and "export to obj" ) option is not available. I've tested with databases that were restored after an app crash (#814) and that were never assembled (app crashes too if I try to "Assemble" -> "Optimize mesh" them).
So if in fact first person view requires an optimized mesh, this is not a bug.

matlabbe, 2023-10-21:
Closing, as FPV only works after assembling the map (if it was not already done in a database we just loaded).

closed 2023-10-21 by matlabbe