introlab/rtabmap #965: Install APK from Android30 Docker fails

pedrojlsilva, 2023-01-27:
Hi guys,

I followed the documentation [step-by-step](https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-arcore-apk) and generated the apk to be installed on a Samsung Galaxy M51 android device. However, when installing, the application is being marked as invalid.

Any ideas on this?

matlabbe, 2023-01-29:
What is your android version on your phone? If you upgraded to Android 12 (API 31-32), that may be why it does want to open it.

nano-meter, 2023-07-12:
I think I figured it out. From API version 30 Android requires a minimum of signature scheme v2.
So this command signs the debug APK:
Run it in the docker container.
```
/opt/android-sdk/build-tools/30.0.3/apksigner sign --ks ~/.android/debug.keystore  rtabmap-tango/build/arm64-v8a/app/android/bin/RTABMap-debug.apk
```
The keystore password is `android`

matlabbe, 2023-07-14:
Thank you, I was trying to install API30 version on my S20+ the other day, and got signature issue, then fallback to API26 apk instead. I added your instructions to [wiki](https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-arcore-apk) for API 30. Thanks!

closed 2023-07-14 by matlabbe