introlab/rtabmap #966: Run error with Ceres

fjlxb, 2023-01-28:
Hey, when run rtabmap standalone, I meet this error:
`Optimizer.cpp:446::optimizeBA() Optimizer 3 doesn't implement optimizeBA() method.`

In Parameters.h, I set Optimizer/Strategy=3, it works

set OdomF2M/BundleAdjustment=3, it meets error, how to fix it? Is anything wrong with Ceres?

matlabbe, 2023-01-29:
Ceres graph optimization is implemented, but not the bundle adjustment part.

fjlxb, 2023-01-29:
> Ceres graph optimization is implemented, ~but not the bundle adjustment part~ EDIT actually it is:
> 
> https://github.com/introlab/rtabmap/blob/128d3b0d0a7a256650288d291c08429d6488a8f5/corelib/src/optimizer/OptimizerCeres.cpp#L316
> 
> 
> looking deeper... ok I see there is indeed a bug, the Optimizer's `optimizeBA` virtual function parameters have been updated and inherited function in Ceres was not. Checking this.

should I update Ceres?

matlabbe, 2023-01-29:
I just fixed the build in this commit: https://github.com/introlab/rtabmap/commit/e4edee33eb8de40a58c02748e2e026e586ed64e8

**However**, ceres will likely fail in odometry if you set `OdomF2M/BundleAdjustment=3`. Only g2o's BA has been tested for odometry. cvsba and ceres were only tested in [step 6 of this tutorial](https://github.com/introlab/rtabmap/wiki/Export-Raster-Layers-to-MeshLab) (in global BA setting, not local BA). Nevertheless, with some time ceres could be fixed for local BA. Code is there: 
https://github.com/introlab/rtabmap/blob/master/corelib/src/optimizer/OptimizerCeres.cpp#L316-L515

matlabbe, 2023-01-29:
Done a quick test based on the ceres errors/warnings, Ceres seems working (`OdomF2M/BundleAdjustment=3`) when changing this line:
https://github.com/introlab/rtabmap/blob/e4edee33eb8de40a58c02748e2e026e586ed64e8/corelib/src/optimizer/OptimizerCeres.cpp#L460
to:
```cpp
	options.linear_solver_type = ceres::ITERATIVE_SCHUR;
```

fjlxb, 2023-01-29:
> Done a quick test based on the ceres errors/warnings, Ceres seems working (`OdomF2M/BundleAdjustment=3`) when changing this line:
> 
> https://github.com/introlab/rtabmap/blob/e4edee33eb8de40a58c02748e2e026e586ed64e8/corelib/src/optimizer/OptimizerCeres.cpp#L460
> 
> 
> to:
> ```c++
> 	options.linear_solver_type = ceres::ITERATIVE_SCHUR;
> ```

Oh, thank you, I have a try then give you reply~

fjlxb, 2023-01-31:
> Done a quick test based on the ceres errors/warnings, Ceres seems working (`OdomF2M/BundleAdjustment=3`) when changing this line: [...]

It works!

matlabbe, 2023-02-01:
I'll try to make it by default, and maybe adding more parameters to Ceres to change the solvers (like I did for gtsam or g2o).

commit a406c377 (2023-03-19): gtsam: Fixed metis linker errors on windows. Ceres: updated default solver options (working for BA and pose-graph optimizations, fixed #966)
closed 2023-03-19 by matlabbe