introlab/rtabmap #978: How to know the coordinate of origin in the pgm?

fjlxb, 2023-02-21:
Hello, 

question is shown in the title,

Thank you~

closed 2023-02-23 by fjlxb

matlabbe, 2023-02-26:
You can get the coordinate of the image corner (-x,-y) with:
```
roscore
rosrun rtabmap_ros rtabmap _database_path:=rtabmap.db
rosrun map_server map_saver map:=grid_map
rosservice call /publish_map 1 1 0
```
For convenience, also updated DbViewer in commit above to export YAML file with origin/resolution when doing File->Export 2d optimized map..

fjlxb, 2023-02-27:
> You can get the coordinate of the image corner (-x,-y) with:您可以通過以下方式獲取圖像角點 (-x,-y) 的坐標：
> 
> ```
> roscore
> rosrun rtabmap_ros rtabmap _database_path:=rtabmap.db
> rosrun map_server map_saver map:=grid_map
> rosservice call /publish_map 1 1 0
> ```
> 
> For convenience, also updated DbViewer in commit above to export YAML file with origin/resolution when doing File->Export 2d optimized map..為了方便起見，還在上面的提交中更新了 DbViewer，以便在執行文件-> 導出 2d 優化地圖時導出帶有原點/分辨率的 YAML 文件。

Thank you~

The yaml file can be got from the newest code.

fjlxb, 2023-02-27:
> You can get the coordinate of the image corner (-x,-y) with:
>
> ```
> roscore
> rosrun rtabmap_ros rtabmap _database_path:=rtabmap.db
> rosrun map_server map_saver map:=grid_map
> rosservice call /publish_map [...]

Thank you~

The yaml file can be got from the newest code.