introlab/rtabmap #980: cmake-ros workflow failing (dpkg: error processing package grub-efi-amd64-signed)

matlabbe, 2023-02-25:
https://github.com/introlab/rtabmap/actions/runs/4250222519/jobs/7436679870

```

  Setting up dotnet-targeting-pack-6.0 (6.0.113-0ubuntu1~22.04.1) ...
  Setting up aspnetcore-targeting-pack-6.0 (6.0.113-0ubuntu1~22.04.1) ...
  Setting up aspnetcore-runtime-6.0 (6.0.113-0ubuntu1~22.04.1) ...
  Setting up dotnet-hostfxr-6.0 (6.0.113-0ubuntu1~22.04.1) ...
  Setting up powershell (7.3.3-1.deb) ...
  Setting up dotnet-sdk-7.0 (7.0.201-1) ...
  Setting up grub-efi-amd64-bin (2.06-2ubuntu14.1) ...
  Setting up grub-efi-amd64-signed (1.187.3~22.04.1+2.06-2ubuntu14.1) ...
  mount: /var/lib/grub/esp: special device /dev/disk/by-id/scsi-14d534654202020200690df035523a242afc173af9e57f2a4-part15 does not exist.
  dpkg: error processing package grub-efi-amd64-signed (--configure):
   installed grub-efi-amd64-signed package post-installation script subprocess returned error exit status 32
  Processing triggers for mailcap (3.70+nmu1ubuntu1) ...
  Processing triggers for hicolor-icon-theme (0.17-2) ...
  Processing triggers for man-db (2.10.2-1) ...
  Errors were encountered while processing:
   grub-efi-amd64-signed
  needrestart is being skipped since dpkg has failed
  E: Sub-process /usr/bin/dpkg returned an error code (1)
Error: The process '/usr/bin/sudo' failed with exit code 100
```

Possible workaround: https://github.com/orgs/community/discussions/47863

matlabbe, 2023-02-25:
Increasing to setup_ros v0.5 doesn't seem to help https://github.com/introlab/rtabmap/actions/runs/4272000374/jobs/7436857616

matlabbe, 2023-02-25:
Other related issues:
https://github.com/ros-tooling/setup-ros/issues/542
https://github.com/ros-tooling/setup-ros/pull/543

christophebedard, 2023-02-26:
> Increasing to setup_ros v0.5

`v0.6` contains a temporary workaround now.