introlab/rtabmap #981: make error with macos

presunshine, 2023-02-28:
When I  type "make -j4" to build RTAB-Map on macos 13.3 , errors occurred as follows:

(base) [~/rtabmap/build]$ make -j4                                                                 [master][ruby-3.0.0]
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryFLOAM.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RegistrationVis.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryMSCKF.cpp.o
In file included from /Users/mac/rtabmap/corelib/src/optimizer/OptimizerGTSAM.cpp:30:
/Users/mac/rtabmap/utilite/src/../include/rtabmap/utilite/UStl.h:189:6: warning: variable 'i' set but not used [-Wunused-but-set-variable]
        int i=0;
            ^
In file included from /Users/mac/rtabmap/corelib/src/odometry/OdometryFLOAM.cpp:33:
/Users/mac/rtabmap/utilite/src/../include/rtabmap/utilite/UStl.h:189:6: warning: variable 'i' set but not used [-Wunused-but-set-variable]
        int i=0;
            ^
In file included from /Users/mac/rtabmap/corelib/src/odometry/OdometryMSCKF.cpp:33:
/Users/mac/rtabmap/utilite/src/../include/rtabmap/utilite/UStl.h:189:6: warning: variable 'i' set but not used [-Wunused-but-set-variable]
        int i=0;
            ^
In file included from /Users/mac/rtabmap/corelib/src/RegistrationVis.cpp:42:
/Users/mac/rtabmap/utilite/src/../include/rtabmap/utilite/UStl.h:189:6: warning: variable 'i' set but not used [-Wunused-but-set-variable]
        int i=0;
            ^
In file included from /Users/mac/rtabmap/corelib/src/RegistrationVis.cpp:51:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/flann.hpp:45:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/algorithms/all_indices.h:36:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/algorithms/kdtree_index.h:48:
/Users/mac/rtabmap/corelib/src/rtflann/util/random.h:120:14: error: no member named 'random_shuffle' in namespace 'std'
        std::random_shuffle(vals_.begin(), vals_.end(), generator);
        ~~~~~^
In file included from /Users/mac/rtabmap/corelib/src/RegistrationVis.cpp:51:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/flann.hpp:45:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/algorithms/all_indices.h:36:
/Users/mac/rtabmap/corelib/src/rtflann/algorithms/kdtree_index.h:680:18: error: no member named 'random_shuffle' in namespace 'std'
            std::random_shuffle(ind.begin(), ind.end());
            ~~~~~^
In file included from /Users/mac/rtabmap/corelib/src/RegistrationVis.cpp:51:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/flann.hpp:45:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/algorithms/all_indices.h:42:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/algorithms/lsh_index.h:49:
/Users/mac/rtabmap/corelib/src/rtflann/util/lsh_table.h:367:10: error: no member named 'random_shuffle' in namespace 'std'
    std::random_shuffle(indices.begin(), indices.end());
    ~~~~~^
In file included from /Users/mac/rtabmap/corelib/src/RegistrationVis.cpp:51:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/flann.hpp:45:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/algorithms/all_indices.h:43:
In file included from /Users/mac/rtabmap/corelib/src/rtflann/algorithms/autotuned_index.h:36:
/Users/mac/rtabmap/corelib/src/rtflann/nn/index_testing.h:253:11: warning: variable 'p1' set but not used [-Wunused-but-set-variable]
    float p1;
          ^
In file included from /Users/mac/rtabmap/corelib/src/optimizer/OptimizerGTSAM.cpp:39:
In file included from /usr/local/include/gtsam/geometry/Pose2.h:23:
In file included from /usr/local/include/gtsam/geometry/BearingRange.h:21:
In file included from /usr/local/include/gtsam/base/Manifold.h:23:
/usr/local/include/gtsam/base/Testable.h:155:26: error: member 'HasTestablePrereqs' cannot have template arguments
    GTSAM_CONCEPT_ASSERT(HasTestablePrereqs<T>);
                         ^                 ~~~
/usr/local/include/gtsam/base/concepts.h:22:45: note: expanded from macro 'GTSAM_CONCEPT_ASSERT'
#define GTSAM_CONCEPT_ASSERT(concept) void*(concept)
                                            ^~~~~~~
In file included from /Users/mac/rtabmap/corelib/src/optimizer/OptimizerGTSAM.cpp:39:
In file included from /usr/local/include/gtsam/geometry/Pose2.h:23:
In file included from /usr/local/include/gtsam/geometry/BearingRange.h:21:
/usr/local/include/gtsam/base/Manifold.h:95:24: error: member 'HasManifoldPrereqs' cannot have template arguments
  GTSAM_CONCEPT_ASSERT(HasManifoldPrereqs<Class>);
                       ^                 ~~~~~~~
/usr/local/include/gtsam/base/concepts.h:22:45: note: expanded from macro 'GTSAM_CONCEPT_ASSERT'
#define GTSAM_CONCEPT_ASSERT(concept) void*(concept)
[... 276 lines cut ...]
/Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk/usr/include/c++/v1/__config:810:49: note: expanded from macro '_LIBCPP_DEPRECATED'
#      define _LIBCPP_DEPRECATED __attribute__((deprecated))
                                                ^
/Users/mac/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:73:32: error: no member named 'make_nvp' in namespace 'boost::serialization'
    ar & boost::serialization::make_nvp("bRef_", const_cast<Unit3&>(bRef_));
         ~~~~~~~~~~~~~~~~~~~~~~^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
1 warning and 20 errors generated.
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
5 warnings and 3 errors generated.
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/RegistrationVis.cpp.o] Error 1
In file included from /Users/mac/rtabmap/corelib/src/odometry/OdometryVINS.cpp:33:
/Users/mac/rtabmap/utilite/src/../include/rtabmap/utilite/UStl.h:189:6: warning: variable 'i' set but not used [-Wunused-but-set-variable]
        int i=0;
            ^
1 warning generated.
In file included from /Users/mac/rtabmap/corelib/src/odometry/OdometryOpenVINS.cpp:33:
/Users/mac/rtabmap/utilite/src/../include/rtabmap/utilite/UStl.h:189:6: warning: variable 'i' set but not used [-Wunused-but-set-variable]
        int i=0;
            ^
1 warning generated.
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
(base) [~/rtabmap/build]$                                                                          [master][ruby-3.0.0]

listout, 2023-05-14:
This is because random_shuffle is deprecated since c++14 and completely removed since c++17

matlabbe, 2023-05-19:
This has been fixed not so long ago: https://github.com/introlab/rtabmap/commit/e3b7e9378da47ea0c5e8fde07a078093c88f7521

Try updating your rtabmap source code.

closed 2023-05-19 by matlabbe