introlab/rtabmap #982: PreUpdateThread performance spike during playback

dimitryn, 2023-02-28:
Hi , 
I am running rtabmap playback dual camera  (enable_visual_odometry_tru:=true use_rvis:=false playback_mode:=true) on ADL x86.
During run there is frames producing spikes in performance , like x3 or x4 average run .
Look like spikes comming from PreUpdateThread thread getting context switch and freeze for ~500 - 600 ms  during update function in Inserting words loop. 
I suspect freeze is during flann addPoints in that loop . 
Any idea , why it can be ?

matlabbe, 2023-03-06:
Yes, it is expected. This happens when the kdtree is re-balanced. This events occur each time the kdtree doubles in size. This is one inconvenience of the incremental dictionary. If the peaks become too long, you may want to switch to a fixed dictionary. To use a fixed dictionary, map the environment one time with default config (incremental), then on another run, set `Kp/DictionaryPath` to that created database and set `Kp/IncrementalDictionary` to false. The peaks won't happen anymore, but you somewhat lose the adaptability of an incremental dictionary.

dimitryn, 2023-03-09:
Hi @matlabbe  also on same issue , I tried to run fixed  as you suggested in localization mode  and get crash
[rgbd_odometry-2] [INFO] [1678373687.459930722] [rtabmap.rgbd_odometry]: Odom: quality=0, std dev=99.995000m|99.995000rad, update time=0.420034s
[rgbd_odometry-2] [FATAL] [1678373688.442636710] [rtabmap.rgbd_odometry]: [FATAL] (2023-03-09 16:54:48.442) VWDictionary.cpp:562::update() Condition (inserted.second) not met!
[rgbd_odometry-2] terminate called after throwing an instance of 'UException'
[rgbd_odometry-2]   what():  [FATAL] (2023-03-09 16:54:48.442) VWDictionary.cpp:562::update() Condition (inserted.second) not met!
[ERROR] [rgbd_odometry-2]: process has died [pid 80545, exit code -6, cmd '/install/rtabmap_ros/lib/rtabmap_ros/rgbd_odometry --Mem/InitWMWithAllNodes false --Mem/IncrementalMemory true --Rtabmap/DetectionRate 8 --Grid/MaxObstacleHeight 0.5 --RGBD/AngularUpdate 0.01 --RGBD/LinearUpdate 0.01 --RGBD/AngularSpeedUpdate 1.0 --RGBD/LinearSpeedUpdate 0.8 --udebug --logfile /log_file --Vis/CorGuessWinSize 0 --ros-args -r __ns:=/rtabmap -p use_sim_time:=True --params-file /tmp/launch_params_6gh7r0bv -r rgb/image:=/camera0/color/image_raw -r depth/image:=/camera0/depth/image_rect_raw -r rgb/camera_info:=/camera0/color/camera_info -r rgbd_image:=rgbd_image -r odom:=visual_odom_topic -r imu:=/imu/data -r rgbd_image0:=/camera0/rgbd_image -r rgbd_image1:=/camera1/rgbd_image -r rgbd_image2:=/camera2/rgbd_image'].

matlabbe, 2023-04-15:
Make sure to not use the fixed dictionary parameter on `rgbd_odometry` node, only for `rtabmap` node.

To avoid rebalancing the tree, you can set parameter `Kp/FlannRebalancingFactor` to 0:
```
rtabmap --params | grep Rebalan
Param: Kp/FlannRebalancingFactor = "2.0"                   [Factor used when rebuilding the incremental FLANN index (see "Kp/IncrementalFlann"). Set <=1 to disable.]
```
I think it may affect time to search over time, but also the quality of the matches to disable rebalancing

dimitryn, 2023-04-17:
Hi @matlabbe  thanks for reply. I rerun our one my records with different NN strategy and kNNBruteForce pruduced max spike in 
 ms twice smaller compare to kNNFlannKdTree . Is there quality difference when running with different strategy ?
Thanks.

matlabbe, 2023-04-20:
I don't have experiments comparing all approaches, so it is why I said "may affect [...] quality".

I am surprised there are spikes with bruteforce, because there should be no any rebalancing (unless the memory copy when dictionary matrix should be resized takes some time), I would expect processing time smoothly increasing indefinitely.

dimitryn, 2023-04-20:
Hi @matlabbe  understood thank you.
on same topic .
understood thank you

I am trying to export dictionary from db file and I am getting VWDictionary.cpp:1468::rtabmap::VWDictionary::exportDictionary() Exporting binary descriptors is not implemented!
What configuration of dictionary (Kp strategy/Detector strategy  ... maybe bytetoFloat ? ) must be  in order to enable this feature ? and what limitation ?

my current setting
     NNStrategy  1
     IncrementalDictionary true
     IncrementalFlann  true
     FlannRebalancingFactor  2.0
     byteToFloat              false
     DetectorStrategy       8
thanks.

dimitryn, 2023-04-23:
@matlabbe  I also manage to separate IncrementalDictionary=false only for rtabmap node , but it still crashing
[rtabmap-5] [FATAL] [1682263397.866134167] [rtabmap.rtabmap]: [FATAL] (2023-04-23 18:23:17.866) Memory.cpp:3401::addLink() Condition (fromS->getWeight() >= 0 && toS->getWeight() >=0) not met!
[rtabmap-5] terminate called after throwing an instance of 'UException'

I notice in log there is twice set fixed dictionary before it crash.

[ INFO] (2023-04-23 18:22:57.533) VWDictionary.cpp:155::setFixedDictionary() incremental=0, oldPath= newPath=/home/administrator/.ros/rtabmap.db, visual words=0
[ WARN] (2023-04-23 18:22:57.533) VWDictionary.cpp:160::setFixedDictionary() Loading fixed vocabulary "/home/administrator/.ros/rtabmap.db", this may

dimitryn, 2023-05-02:
Hi @matlabbe  , I manage to execute fixed dictionary mode (export/import)
one small quastion dumpDictionary functions saving 2 files DumpMemoryWordRef and DumpMemoryWordDesc. on import only DumpMemoryWordDesc is used .where/when  DumpMemoryWordRef in use ?
thanks.

matlabbe, 2023-05-14:
`exportDictionary` is quite an old feature, even before binary descriptors were added. To support binary descriptors, I would change
https://github.com/introlab/rtabmap/blob/682d54725ac4c3da231c4c9307da3b9004dd2506/corelib/src/VWDictionary.cpp#L1516-L1529
to something like:
```cpp
const unsigned char * desc = (const unsigned char *)(*iter).second->getDescriptor().data;
int dim = (*iter).second->getDescriptor().cols;

for(int i=0; i<dim; i++)
{
	fprintf(foutDesc, "%d ", (int)desc[i]);
}
fprintf(foutDesc, "\n");
```
One problem is if you want to import back the file, it won't work because of 
https://github.com/introlab/rtabmap/blob/682d54725ac4c3da231c4c9307da3b9004dd2506/corelib/src/VWDictionary.cpp#L223-L227

I think the file format would need to be updated to include the feature type (float or binary). Note that I prefer using database format for fixed dictionary instead, which supports both float and binary descriptors.

On import, we don't re-load DumpMemoryWordRef because we don't need it (it is exported just for debugging).

dimitryn, 2023-05-14:
@matlabbe thanks

closed 2023-05-14 by dimitryn

hellovuong, 2024-02-13:
> Yes, it is expected. This happens when the kdtree is re-balanced. This events occur each time the kdtree doubles in size. This is one inconvenience of the incremental dictionary. If the peaks become [...]

Hello @matlabbe Sorry for replying in closed issue. I'd like to know how to achieve your idea when using rtabmap_ros. Can I do it as follows:

1. set_mode_mapping with increment dictionary to true.
2. set_mode_localization to save the map
3. set_mode_mapping again with the fixed dictionary to true and set back to localization immediately.
Can you suggest the steps above or is there a better way to do it?
Thank you!

matlabbe, 2024-02-17:
Based on instructions on this page: https://github.com/introlab/rtabmap/issues/942#issuecomment-1345657263

For ROS it would be translated to :
1. Make a new map with default parameters of a small part of the environment representative enough of the whole target environment (e.g., that would include most of the features that the robot could see in the whole environment)
    * Advanced users could set `Mem/BinDataKept=false` to just keep the visual words (smaller database).
3. Shutdown rtabmap node to save the database (default it is at `~/.ros/rtabmap.db`), and rename it to `~/dictionary.db`
4. In your launch file, add
   ```xml
   <param name="Kp/IncrementalDictionary" value="false"/> 
   <param name="Kp/DictionaryPath" value="~/dictionary.db"/>
   ```
5. Relaunch the launch file, you are now using the fixed dictionary.