introlab/rtabmap #984: make failed, g2o issue

cu-autobike, 2023-03-05:
I followed the instructions on https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817 to install rtabmap on ros-melodic for the Jetson Xavier (running Ubuntu 18.04), but encountered the following error message when running `make`:
`/home/autobike/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:157:133: error: no matching function for call to ‘g2o::Factory::registerType(const char [17], std::unique_ptr<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset>, std::default_delete<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> > >)’ erType("CACHE_SE3_OFFSET", g2o::make_unique<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> >());
In file included from /home/autobike/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:46:0:
/opt/ros/melodic/include/g2o/core/factory.h:62:12: note: candidate: void g2o::Factory::registerType(const string&, g2o::AbstractHyperGraphElementCreator*)
       void registerType(const std::string& tag, AbstractHyperGraphElementCreator* c);
/opt/ros/melodic/include/g2o/core/factory.h:62:12: note:   no known conversion for argument 2 from ‘std::unique_ptr<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset>, std::default_delete<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> > >’ to ‘g2o::AbstractHyperGraphElementCreator*’ corelib/src/CMakeFiles/rtabmap_core.dir/build.make:915: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o] Error 1
CMakeFiles/Makefile2:825: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2`

The cmake output can be found here 
[cmakeoutput.txt](https://github.com/introlab/rtabmap/files/10892371/cmakeoutput.txt)

matlabbe, 2023-03-05:
cmake is mixing up two g2o. In cmake log, it found the one you built locally in `/usr/local`, but in that log error, it is referring to a g2o file in `/opt/ros/melodic`. If you want to use your local version, uninstall `ros-melodic-libg2o`, cleanup the build cache and rebuild.

cheers,
Mathieu

cu-autobike, 2023-03-09:
Thanks, this fixed the issue!

closed 2023-03-09 by cu-autobike