introlab/rtabmap #985: Using Depth->Scan for Lidar based algorithms

alexk1976, 2023-03-07:
Hi Mathieu,
Our setup has 4 RGB-D (~6 meters range) cameras covering almost 360 FOV.
I explore ways to improve loop-detection accuracy that is not perfect now (sometimes there are wrong loop-closures/sometimes no loop-closure where expected).
Whether make sense to convert depth to scan data and use Lidar rtabmap capabilities on top of VSLAM/VO rtabmap algos. 
Particularly proximity detection in loop-closures?
Any other recommendations ? 
thank you

matlabbe, 2023-04-15:
I don't think simulating a fake lidar from the camera will increase loop closure accuracy/precision.

> wrong loop-closures/sometimes

There are some ways to decrease wrong loop closures to be accepted, like increasing `Vis/MinInliers` or decreasing `RGBD/OptimizeMaxError`.

dStanhill, 2023-07-05:
Hi Mathieu,
Further pursuing Alex's direction - if we do want to check if fake lidar using depthimage_to_laserscan can help us in cases we have multiple cameras covering close to 360deg. 
Is multi-camera (=> multi scans) supported ?  How do you suggest going about it, merging the multiple scanlines to one?
Thanks, David

matlabbe, 2023-07-10:
Yes, if you have multiple cameras to increase FOV, a resulting fake >180 deg FOV laser scan could help with loop closures.
 
You may use a node to convert the laser scan to point cloud, then assemble the points clouds using [point_cloud_aggregator](http://wiki.ros.org/rtabmap_util#rtabmap_util.2Fpoint_cloud_aggregator) with a fixed frame like wheel odometry to adjust frames taken a different time. Feed the combined cloud to rtabmap or convert it back to a laser scan so that rtabmap knows it is a 2D laser scan, not arbitrary 3d point cloud. For loop closure it wont change anything, but for 2D occupancy grid ray tracing, it would be faster in 2D.