introlab/rtabmap #986: code examples

nesquik011, 2023-03-10:
hello , i want to know where is the code examples please ? i want stereo mapping , slam etc , i only saw few examples where is the rest ? and i wont use ros

matlabbe, 2023-03-10:
For the standalone library:
 * Examples: https://github.com/introlab/rtabmap/tree/master/examples
 * Tutorials: https://github.com/introlab/rtabmap/wiki/Tutorials

If you want to do SLAM on a robot (or use a Lidar), I strongly recommend to use [rtabmap_ros](https://github.com/introlab/rtabmap_ros) package. The examples with the standalone library can only do single camera VSLAM.

nesquik011, 2023-03-11:
i am new to this , i cant find C++ code examples  for stereo vision SLAM

matlabbe, 2023-04-15:
This example works also for stereo: https://github.com/introlab/rtabmap/blob/master/examples/RGBDMapping/main.cpp

It depends which kind of stereo camera you have. If this a supported camera in rtabmap, you may use that example directly.