introlab/rtabmap #995: vcpkg: error when building Debug config with octomap dependency

matlabbe, 2023-03-18:
```
octomap.lib(OcTreeNode.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match
 value '2' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomap.lib(OcTreeNode.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' does
n't match value 'MDd_DynamicDebug' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomap.lib(Pointcloud.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match
 value '2' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomap.lib(Pointcloud.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' does
n't match value 'MDd_DynamicDebug' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomap.lib(AbstractOcTree.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't m
atch value '2' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomap.lib(AbstractOcTree.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease'
doesn't match value 'MDd_DynamicDebug' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomap.lib(AbstractOccupancyOcTree.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0'
doesn't match value '2' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomap.lib(AbstractOccupancyOcTree.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_Dynamic
Release' doesn't match value 'MDd_DynamicDebug' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcx
proj]
octomap.lib(ColorOcTree.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't matc
h value '2' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomap.lib(ColorOcTree.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doe
sn't match value 'MDd_DynamicDebug' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomath.lib(Vector3.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match v
alue '2' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomath.lib(Vector3.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn'
t match value 'MDd_DynamicDebug' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomath.lib(Pose6D.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match va
lue '2' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomath.lib(Pose6D.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't
 match value 'MDd_DynamicDebug' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomath.lib(Quaternion.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't matc
h value '2' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
octomath.lib(Quaternion.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doe
sn't match value 'MDd_DynamicDebug' in Rtabmap.obj [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
```

Neumann-A, 2023-03-18:
Do you use `OCTOMAP_LIBRARIES` for linkage? If yes use the target `octomap` instead

matlabbe, 2023-03-18:
This issue is fixed using target `octomap`, thx @Neumann-A to suggest that. However, I got this new error:
```
OctoMap.obj : error LNK2001: unresolved external symbol "protected: static class rtabmap::RtabmapColorOcTree::StaticMem
berInitializer rtabmap::RtabmapColorOcTree::RtabmapColorOcTreeMemberInit" (?RtabmapColorOcTreeMemberInit@RtabmapColorOc
Tree@rtabmap@@1VStaticMemberInitializer@12@A) [C:\workspace\rtabmap\build\corelib\src\rtabmap_core.vcxproj]
C:\workspace\rtabmap\build\bin\rtabmap_cored.dll : fatal error LNK1120: 1 unresolved externals [C:\workspace\rtabmap\bu
ild\corelib\src\rtabmap_core.vcxproj]
```

This is related to this: 
https://github.com/introlab/rtabmap/blob/ab99719a78de5ffe6dd9f22576eed3f56a3aa731/corelib/src/OctoMap.cpp#L281-L284

Removing the `#ifndef` makes still freezing RTAB-Map on start, but avoid the compilation error.

matlabbe, 2023-03-18:
Closing as original error is fixed, but closing as WONT FIX for the other error.

closed 2023-03-18 by matlabbe

Neumann-A, 2023-03-18:
> still freeze RTAB-Map on start (like the comment says)

Hook up a debugger and see where it freezes? (either let VS start the app or just attach a debugger to it.)

matlabbe, 2023-03-18:
Just tried launching with visual studio debugger, when pausing it seems stuck in OpenMP stuff:
```
#pragma omp parallel
```
[image]

Neumann-A, 2023-03-18:
I don't feel like the openmp directives make sense there. (https://learn.microsoft.com/en-us/cpp/parallel/openmp/reference/openmp-directives?view=msvc-170)

matlabbe, 2023-03-18:
This code is from OctoMap library: 
https://github.com/OctoMap/octomap/blob/92a9f14cafa5f567c47b58cbe91137c9a6ba1b0c/octomap/include/octomap/OcTreeBaseImpl.hxx#L96-L109