introlab/rtabmap #998: Cannnot build iOS. 'rtabmap/utilite/ULogger.h' file not found

Woojin-Crive, 2023-03-21:
I am trying to build iOS app and run on iPhone 13 pro max.

I used /app/ios/RTABMapApp/install_deps.sh to install dependency.

After, i changed signing team and bundle identifier. I used xcode's play button to build.

[image]

But  'rtabmap/utilite/ULogger.h' file not found message is displayed.

How can i handle this problem?

Genozen, 2023-03-23:
In your build settings, search `search header path` in there you can specify where the path of your headers are (I recommend just give the path of the entire rtabmap folder) then on the right side select `recursive` search. This will solve most of your header not found issues.

matlabbe, 2023-03-25:
The problem is that the minor version has been updated (`0.21`) and the headers were installed in `rtabmap-0.21` instead of `rtabmap-0.20`. I updated the Xcode project in commit above, but in the future if it happens again, update the path here:
[image]

closed 2023-03-25 by matlabbe