introlab/rtabmap #999: rtabmap/build make failed

ParkJS99, 2023-03-22:
```
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake ..
**$ make -j4**
$ sudo make install
```

when i'm trying to download project( [from this guide ](https://github.com/introlab/rtabmap/wiki/Installation#ubuntu))

error appears like this

```
'corelib/src/CMakeFiles/rtabmap_core.dir/build.make:915: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o' failed'
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o] Error 1
CMakeFiles/Makefile2:825: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2
```

[image]

I think there is a problem with g2o and "make_unique" seems to be the keyword for the error
But I don't know how to solve it.

OS : Ubuntu 18.04
opencv : 4.4.0
cmake : 3.25

ParkJS99, 2023-03-23:
> Is g2o built from source? (Which version?) Or is it installed by ros-melodic-libg2o?

1. yes i did. (lastest version, C++17)
👇 what i tried
'''
$ sudo apt-get update
$ sudo apt-get install libsqlite3-dev libpcl-dev libopencv-dev git cmake libproj-dev libqt5svg5-dev

$ cd opencv-4.4.0
$ mkdir build
$ cd build
$ cmake -DOPENCV_EXTRA_MODULES_PATH=~/opencv/opencv_contrib-4.4.0/modules -D CMAKE_BUILD_TYPE=RELEASE -D CMAKE_INSTALL_PREFIX=/usr/local -D WITH_TBB=ON -D BUILD_NEW_PYTHON_SUPPORT=ON -D WITH_V4L=ON -D INSTALL_C_EXAMPLES=ON -D INSTALL_PYTHON_EXAMPLES=ON -D BUILD_EXAMPLES=ON -D WITH_QT=ON -D WITH_OPENGL=ON -D BUILD_PERF_TESTS=OFF -D BUILD_TESTS=OFF -D BUILD_LIBPROTOBUF_FROM_SOURCES=ON -D OPENCV_GENERATE_PKGCONFIG=ON ..
$ make -j6
$ sudo make install

#G2O
$ git clone https://github.com/RainerKuemmerle/g2o.git 
$ cd g2o
$ mkdir build
$ cd build
$ cmake -DBUILD_WITH_MARCH_NATIVE=OFF -DG2O_BUILD_APPS=OFF -DG2O_BUILD_EXAMPLES=OFF -DG2O_USE_OPENGL=OFF ..
$ make -j4
$ sudo make install

#GTSAM
$ sudo add-apt-repository ppa:borglab/gtsam-release-4.0
$ sudo apt update 
$ sudo apt install libgtsam-dev libgtsam-unstable-dev

#RTAB
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake ..
$ make -j4
$ sudo make install
'''

2. never used or install the ros-melodic-libg2o, so maybe not.

ETC. gcc version : gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

matlabbe, 2023-03-23:
Humm, g2o just updated to c++17 4 days ago: https://github.com/RainerKuemmerle/g2o/commit/485c2a39755255e4911be8e678bb0541f0511efc

I'll check to update the interface on rtabmap side.

matlabbe, 2023-03-23:
The commit above should fix the issue.

closed 2023-03-23 by matlabbe

ParkJS99, 2023-03-23:
> The commit above should fix the issue.

sorry but it's not working though  

even shows more error like GTSAM failed
:cry:

[image]

ParkJS99, 2023-03-23:
> ```
> $ git clone https://github.com/introlab/rtabmap.git rtabmap
> $ cd rtabmap/build
> $ cmake ..
> **$ make -j4**
> $ sudo make install
> ```
>
> when i'm trying to download project( [from this guide [...]

ParkJS99, 2023-03-23:
i use this g2o
[e8df200](https://github.com/RainerKuemmerle/g2o/commit/e8df2004e07ea8f5b8e6a8b9f2dc067b45b45036)
version and G2O Error is not appears now.

but still GTSAM Error occurred

matlabbe, 2023-03-24:
Yeah, I didn't take the time yet to make gtsam > 4.0 working with rtabmap. The latest errors you have are related to opencv. It seems that xfeatures2d module of opencv has been detected, but the library is not correctly linked. I'll suggest to restart a clean build (remove `CMakeCache.txt` or anything in build directory).

ParkJS99, 2023-03-24:
Thank you for response,
solved cv error but i'm stuck on pcl(?) error now.

[image]

Thank you for your kindness.

ETC.

```
-- Info :
--   RTAB-Map Version =     0.21.0
--   CMAKE_VERSION =        3.25.2
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++17
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS =  ;-DDISABLE_ENSENSO;-DDISABLE_DAVIDSDK;-DDISABLE_DSSDK;-DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0;-DFLANN_STATIC;-Dqh_QHpointer;-DDISABLE_RSSDK
--   PCL_VERSION = 1.8.1
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4.4.0 xfeatures2d = YES, nonfree = NO (License: BSD)
--   With Qt 5.9.7            = YES (License: Open Source or Commercial)
--   With VTK 6.3              = YES (License: BSD)
--   With external SQLite3     = YES (License: Public Domain)
--   With ORB OcTree           = YES (License: GPLv3)
--   With SupertPoint          = NO (WITH_TORCH=OFF)
--   With Python3              = NO (WITH_PYTHON=OFF)
--   With Madgwick             = YES (License: GPL)
--   With FastCV               = NO (FastCV not found)
--   With PDAL                 = NO (PDAL not found)
-- 
--  Solvers:
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--  *With g2o 1.0.0            = YES (License: BSD)
--  *With GTSAM 4.0.3          = YES (License: BSD)
--  *With Ceres                = NO (WITH_CERES=OFF)
--   With MRPT                 = NO (MRPT not found)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (WITH_CVSBA=OFF)
--  *With libpointmatcher      = NO (libpointmatcher not found)
--   With CCCoreLib            = NO (WITH_CCCORELIB=OFF)
--   With Open3D               = NO (WITH_OPEN3D=OFF)
--   With OpenGV               = NO (WITH_OPENGV=OFF)
-- 
--  Reconstruction Approaches:
--   With OCTOMAP              = NO (octomap not found)
--   With CPUTSDF              = NO (WITH_CPUTSDF=OFF)
--   With OpenChisel           = NO (WITH_OPENCHISEL=OFF)
--   With AliceVision          = NO (WITH_ALICE_VISION=OFF)
-- 
--  Camera Drivers:
--   With Freenect             = NO (libfreenect not found)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
```

matlabbe, 2023-03-24:
The anaconda envrionemnt has some shared libraries with different versions than in the system. After changing anaconda path to avoid linker to find libraries in it, delete everything in rtabmap build directory to do a fresh build. Even if you change some paths after build, some stay in CMakCache.txt, which may cause problems.

an99990, 2023-04-09:
i noticed that in the PR theres a /* [missing at the beginning ](https://github.com/introlab/rtabmap/commit/85cc6fe3c742855ad16c8442895e12dbb10b6e8b#:~:text=%40%40%20%2D1%2C4%20%2B1%2C3%20%40%40-,/*,-Copyright%20(c)%202010)and thats what gave issue while building
https://github.com/introlab/rtabmap/commit/85cc6fe3c742855ad16c8442895e12dbb10b6e8b#:~:text=%40%40%20%2D1%2C4%20%2B1%2C3%20%40%40-,/*,-Copyright%20(c)%202010

matlabbe, 2023-04-09:
It was added back in https://github.com/introlab/rtabmap/commit/c4e94bcdc31b859c1049724dbb7671e4597d86de