introlab/rtabmap #1000: make failed, gtsam issue

cu-autobike, 2023-03-23:
I followed the instructions on [#427 (comment)](https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817) to install rtabmap on ros-melodic for the Jetson Xavier (running Ubuntu 18.04), but encountered the following error message when running make:

`In file included from /usr/include/gtsam/slam/PriorFactor.h:20:0,
                 from /home/autobike/rtabmap/corelib/src/optimizer/OptimizerGTSAM.cpp:43:
/usr/include/gtsam/nonlinear/PriorFactor.h: In instantiation of ‘gtsam::Vector gtsam::PriorFactor<VALUE>::evaluateError(const T&, gtsam::OptionalMatrixType) const [with VALUE = vertigo::SwitchVariableLinear; gtsam::Vector = Eigen::Matrix<double, -1, 1>; gtsam::PriorFactor<VALUE>::T = vertigo::SwitchVariableLinear; gtsam::OptionalMatrixType = Eigen::Matrix<double, -1, -1>*; gtsam::Matrix = Eigen::Matrix<double, -1, -1>]’:
/home/autobike/rtabmap/corelib/src/optimizer/OptimizerGTSAM.cpp:749:1:   required from here
/usr/include/gtsam/nonlinear/PriorFactor.h:101:31: error: call to ‘static gtsam::traits<vertigo::SwitchVariableLinear>::TangentVector gtsam::traits<vertigo::SwitchVariableLinear>::Local(const vertigo::SwitchVariableLinear&, const vertigo::SwitchVariableLinear&, gtsam::traits<vertigo::SwitchVariableLinear>::ChartJacobian, gtsam::traits<vertigo::SwitchVariableLinear>::ChartJacobian)’ uses the default argument for parameter 3, which is not yet defined
       return -traits<T>::Local(x, prior_);
               ~~~~~~~~~~~~~~~~^~~~~~~~~~~
/usr/include/gtsam/nonlinear/PriorFactor.h:101:31: error: call to ‘static gtsam::traits<vertigo::SwitchVariableLinear>::TangentVector gtsam::traits<vertigo::SwitchVariableLinear>::Local(const vertigo::SwitchVariableLinear&, const vertigo::SwitchVariableLinear&, gtsam::traits<vertigo::SwitchVariableLinear>::ChartJacobian, gtsam::traits<vertigo::SwitchVariableLinear>::ChartJacobian)’ uses the default argument for parameter 4, which is not yet defined
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:929: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o] Error 1
CMakeFiles/Makefile2:825: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2`

The cmake output can be found here:
[cmakeoutput2.txt](https://github.com/introlab/rtabmap/files/11045902/cmakeoutput2.txt)

matlabbe, 2023-03-23:
It is gtsam 4.3. rtabmap is currently tested only with gtsam 4.0

matlabbe, 2023-05-12:
With latest gtsam:
```
#0 42.06 In file included from /root/rtabmap/corelib/src/optimizer/OptimizerGTSAM.cpp:53:
#0 42.06 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:66:40: error: ‘none’ is not a member of ‘boost’
#0 42.06    66 |       OptionalJacobian<2,3> H = boost::none) const;
#0 42.06       |                                        ^~~~
#0 42.06 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h: In member function ‘void rtabmap::GravityFactor::serialize(ARCHIVE&, unsigned int)’:
#0 42.06 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:72:32: error: ‘make_nvp’ is not a member of ‘boost::serialization’
#0 42.06    72 |     ar & boost::serialization::make_nvp("nZ_", const_cast<Unit3&>(nZ_));
#0 42.06       |                                ^~~~~~~~
#0 42.06 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:73:32: error: ‘make_nvp’ is not a member of ‘boost::serialization’
#0 42.06    73 |     ar & boost::serialization::make_nvp("bRef_", const_cast<Unit3&>(bRef_));
#0 42.06       |                                ^~~~~~~~
#0 42.06 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h: At global scope:
#0 42.06 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:88:18: error: ‘shared_ptr’ in namespace ‘boost’ does not name a template type
#0 42.06    88 |   typedef boost::shared_ptr<Rot3GravityFactor> shared_ptr;
#0 42.06       |                  ^~~~~~~~~~
#0 42.06 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:127:7: error: ‘boost::optional’ has not been declared
#0 42.06   127 |       boost::optional<Matrix&> H = boost::none) const {
#0 42.06       |       ^~~~~
#0 42.06 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:127:22: error: expected ‘,’ or ‘...’ before ‘<’ token
#0 42.06   127 |       boost::optional<Matrix&> H = boost::none) const {
#0 42.06       |                      ^
#0 42.07 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h: In member function ‘virtual gtsam::NonlinearFactor::shared_ptr rtabmap::Rot3GravityFactor::clone() const’:
#0 42.07 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:114:19: error: ‘static_pointer_cast’ is not a member of ‘boost’
#0 42.07   114 |     return boost::static_pointer_cast<gtsam::NonlinearFactor>(
#0 42.07       |                   ^~~~~~~~~~~~~~~~~~~
#0 42.07 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:114:19: note: suggested alternatives:
#0 42.07 In file included from /usr/include/c++/11/memory:77,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:11,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core/cvstd.hpp:81,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core/base.hpp:58,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core.hpp:53,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core/core.hpp:48,
#0 42.07                  from /root/rtabmap/corelib/src/../include/rtabmap/core/Transform.h:37,
#0 42.07                  from /root/rtabmap/corelib/src/../include/rtabmap/core/Link.h:33,
#0 42.07                  from /root/rtabmap/corelib/src/../include/rtabmap/core/Graph.h:36,
#0 42.07                  from /root/rtabmap/corelib/src/optimizer/OptimizerGTSAM.cpp:27:
#0 42.07 /usr/include/c++/11/bits/shared_ptr.h:582:5: note:   ‘std::static_pointer_cast’
#0 42.07   582 |     static_pointer_cast(const shared_ptr<_Up>& __r) noexcept
#0 42.07       |     ^~~~~~~~~~~~~~~~~~~
#0 42.07 In file included from /usr/include/c++/11/bits/shared_ptr.h:53,
#0 42.07                  from /usr/include/c++/11/memory:77,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:11,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core/cvstd.hpp:81,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core/base.hpp:58,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core.hpp:53,
#0 42.07                  from /usr/local/include/opencv4/opencv2/core/core.hpp:48,
#0 42.07                  from /root/rtabmap/corelib/src/../include/rtabmap/core/Transform.h:37,
#0 42.07                  from /root/rtabmap/corelib/src/../include/rtabmap/core/Link.h:33,
#0 42.07                  from /root/rtabmap/corelib/src/../include/rtabmap/core/Graph.h:36,
#0 42.07                  from /root/rtabmap/corelib/src/optimizer/OptimizerGTSAM.cpp:27:
#0 42.07 /usr/include/c++/11/bits/shared_ptr_base.h:1546:5: note:   ‘std::static_pointer_cast’
#0 42.07  1546 |     static_pointer_cast(const __shared_ptr<_Tp1, _Lp>& __r) noexcept
#0 42.07       |     ^~~~~~~~~~~~~~~~~~~
#0 42.07 In file included from /root/rtabmap/corelib/src/optimizer/OptimizerGTSAM.cpp:53:
#0 42.07 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:114:61: error: expected primary-expression before ‘>’ token
#0 42.07   114 |     return boost::static_pointer_cast<gtsam::NonlinearFactor>(
#0 42.07       |                                                             ^
#0 42.07 /root/rtabmap/corelib/src/optimizer/gtsam/GravityFactor.h:115:58: error: invalid new-expression of abstract class type ‘rtabmap::Rot3GravityFactor’
#0 42.07   115 |         gtsam::NonlinearFactor::shared_ptr(new This(*this)));
#0 42.07       |                                                          ^
[... cut ...]
#0 50.46 [ 31%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryOpen3D.cpp.o
#0 52.47 [ 31%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/IMU.cpp.o
#0 53.14 make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:930: corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o] Error 1
#0 53.14 make[2]: *** Waiting for unfinished jobs....
#0 67.61 make[1]: *** [CMakeFiles/Makefile2:826: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
#0 67.61 make: *** [Makefile:156: all] Error 2
```

closed 2023-05-14 by matlabbe