introlab/rtabmap #1001: iOS crash due to reproject error

Genozen, 2023-03-23:
Spec:
iOS rtabmap version 0.20.19
Iphone 14 iOS 16.2

Error:
```
terminating with uncaught exception of type UException: [FATAL] {8126856192} (2023-03-21 14:21:00.002) CameraModel.cpp:790::reproject() Condition (z!=0.0f) not met!
```

Upon inspection of the code of `CameraModel.cpp` we noticed the line at 790 `UASSERT(z!=0.0f);` would cause the app to crash instantly.

I have verified that the RAM was still okay when reviewing similar issue #814 , the ram was plentiful and crashes at random range 600 MB ~ 2000 MB.

Currently trying to do a catch on z == 0.0f and assign it to 1 to see if this would fix the crash, but mathematically this might not be the best way to assign values as it handles the reprojection.

matlabbe, 2023-03-23:
That function would be called from here: https://github.com/introlab/rtabmap/blob/1173f06bc876b22e796770cb49eb1a4a73c118e2/app/android/jni/CameraMobile.cpp#L436

You may try to disable "Save raw scans" in Settings->Mapping...

Genozen, 2023-03-23:
> That function would be called from here:
> 
> https://github.com/introlab/rtabmap/blob/1173f06bc876b22e796770cb49eb1a4a73c118e2/app/android/jni/CameraMobile.cpp#L436
> 
> You may try to disable "Save raw scans" in Settings->Mapping...

Hi @matlabbe , thanks for the swift reply!

I will try it with the disabled settings. And to confirm are the reprojected points being used anywhere else? Such as storing it in a pose graph for loop closure, or Odometry estimations? (just from glancing at the code, I have not seen anything too obvious).

matlabbe, 2023-03-24:
Well, looking in more details, that option does nothing on RTAB-Map iOS. It seems the UI option has been added from android but the functionality is not implemented, thus that option should be removed.

With iOS device having LiDAR, that option is more for debugging purpose. Features tracked by ARKit will be saved as laser scan in each node. See https://developer.apple.com/documentation/arkit/arframe/2887449-rawfeaturepoints for description.

If you switch off "LiDAR" mode, or you have a iPhone without Lidar, the ARKit features will be used for loop closure detection and generate a rough point cloud of the scene.

The app is crashing there because a feature tracked by ARKit seems returning Z=0. I didn't know it was possible, so I'll make check to avoid asserting, by just ignoring that point.

matlabbe, 2023-03-25:
New iOS version 0.21.0 currently under review on App Store, I added a workaround in commit above to avoid the assert.

closed 2023-03-25 by matlabbe

Genozen, 2023-04-04:
> 

@matlabbe  We can help test it out once the update is approved. So far we added our own to ignore this part and it has not crashed 😀