introlab/rtabmap #1006: A quick question about the detection rate

takahiko-hasegawa-hdjp, 2023-03-30:
I noticed that there is Rtabmap/DetectionRate parameter to change the frequency, but what is the point of having this? 
I assume making that to "0" would make it as fast as possible, but what is the point of decreasing the frequency here?

matlabbe, 2023-04-02:
I don't recommend to set to 0 unless you are reprocessing a database offline and make sure that all frames are processed.

`0.5` would be a map update each 2 seconds. With some memory/cpu constraints, you may want the map to increase in size slower over time to map larger environment with same memory usage. 

On most applications, I don't see the need to increase it more than 1 Hz, otherwise it would be a waste of memory and computation time.

takahiko-hasegawa-hdjp, 2023-04-02:
In my case, the input image is coming in at 10Hz. I've been trying to create a map usign default deteactionrate value but when i changed that to 0, the odometry looked better but yes it does use a lot of CPU. 

Thanks for clarification

closed 2023-04-16 by matlabbe