introlab/rtabmap #1007: Fusing gps data with visual data

takahiko-hasegawa-hdjp, 2023-04-03:
I understand we input gps data to rtabmap. But what is the logic behind it?

matlabbe, 2023-04-16:
See https://answers.ros.org/question/329165/using-rtabmap_ros-with-external-gps/?answer=329205#post-id-329205

closed 2023-04-16 by matlabbe

peta-peta, 2023-04-16:
Is there a reason why GPS support is not used directly in rtabmap standalone version? Do you want to keep it relying to visual data purposely? Just a question :)

matlabbe, 2023-04-17:
It is supproted in android and iOS apps. For the standalone version, it is just we didn't implement any GPS drivers. Like for LiDARs, external IMUs or external odometry, we put more effort on support them under ROS. For more custom setups, I suggest to use `rtabmap_ros`. However, if there is a standard library somewhere to get GPS data from many GPS devices, we could integrate it in the standalone.

peta-peta, 2023-04-17:
Thanks for your clear answer. Understood.