introlab/rtabmap #1014: Mapping/Map optimizing getting stuck when detecting markers

kojokaro, 2023-04-13:
Hi

I'm trying to enable markers detection in rtabmap.
I enabled MarkerDetection and set Marker/Length to 0.15.
[log.markers.txt](https://github.com/introlab/rtabmap/files/11222612/log.markers.txt)
[log.nomarkers.txt](https://github.com/introlab/rtabmap/files/11222614/log.nomarkers.txt)
[log.markers.txt](https://github.com/introlab/rtabmap/files/11222664/log.markers.txt)
[log.nomarkers.txt](https://github.com/introlab/rtabmap/files/11222665/log.nomarkers.txt)

I see that a marker is detected on the first frame, but then I never get to Rtabmap:Process() function again. Looks like it's stuck in some recursion around FlannIndex::addPoints() / VWDictionary::update() / RegistrationVis::computeTransformationImpl().

When I hacked MarkerDetector::detect() function to clear the markers detected vector (as if no marker was detected), I see many calls to Rtabmap::Process() after the first frame, so it seems that somehow the detection of the marker causes infinite recursion.

Any idea what it can be?
Attaching 2 log files:
log.markers.txt - when the marker is detected and we get stuck
log.nomarkers.txt - with the hack (no detection), next frames are processed

Thanks!

matlabbe, 2023-04-17:
The main process looks stuck somewhere. During that time the odometry thread is still running in the background, so it is not a recursive thing, it is normal. The issue is that rtabmap node looks stuck. However, I don't know if it is just that log output doesn't appear in time or there is a lag (on ROS2 depending of the config, the logs won't be shown for a while, then show up at once at some point with some latency). 

Can you do `ros2 topic hz` on the output `/rtabmap/info` topic? to see if there is indeed a lag in processing and not just a logging issue.

kojokaro, 2023-04-18:
Thanks Mathieu.

It seems that it was just my mistake.
Closing the ticket.

closed 2023-04-18 by kojokaro