introlab/rtabmap #1019: Fails to launch on M1 Ventura

JetForMe, 2023-04-18:
macOS gatekeeper says it can't check the app for malicious code. Forcing the launch leads to it crashing due to a missing library. I guess dependencies need to be installed in addition to running the `.app`?

```
Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /opt/homebrew/*/libpdal_util.15.dylib
Referenced from: <182225D2-B2A3-3212-87F9-10AE0E18AD5E> /Applications/Development/RTABMap.app/Contents/Frameworks/libpdalcpp.15.2.0.dylib
Reason: tried: '/opt/homebrew/*/libpdal_util.15.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/*/libpdal_util.15.dylib' (no such file), '/opt/homebrew/*/libpdal_util.15.dylib' (no such file), '/usr/local/lib/libpdal_util.15.dylib' (no such file), '/usr/lib/libpdal_util.15.dylib' (no such file, not in dyld cache)
```

matlabbe, 2023-04-20:
The [RTABMap.app](https://github.com/introlab/rtabmap/releases/download/0.21.0/RTABMap-0.21.0-Darwin-arm64.dmg) should contain all dependencies, so it is a bug if it is failing because of missing library error. I'll try to check this soon.

For 
> macOS gatekeeper says it can't check the app for malicious code

as this is not distributed within the App Store, MacOS will always complain about this. The app can be unlocked in Settings->Security.

JetForMe, 2023-04-20:
I would've thought it should be able to check for malicious code, just not verify the developer signature.

matlabbe, 2023-04-21:
I updated the release to fix the missing `libpdal...`issue: https://github.com/introlab/rtabmap/releases/tag/0.21.0

About malicious code detection, yeah I am not sure how Mac OS works too.

closed 2023-04-21 by matlabbe

JetForMe, 2023-04-25:
Seems to launch now, thanks!