introlab/rtabmap #1020: Save new mapping session to seperate database file

hellovuong, 2023-04-19:
Hello, 
I am wondering if is there any way that allows us to save the new mapping session into a new db file after we finish this session.
For more detail, I had a database file and load it for localization. After a while, I start a new mapping session again to update the database. After finishing mapping, the new session of the course will save in the existing database file, also I would like to save it to a separate file. Could give me some idea how to do it?
Thank you,

matlabbe, 2023-04-20:
The easiest would be to use `rtabmap-reprocess` tool, starting from last session:
```
rtabmap-reprocess -start_s 2 rtabmap.db output.db
```
This will make sure that the visual dictionary is cleanly rebuilt with data only from latest session. 

You could get latest session id from the db with:
```
sqlite3 rtabmap.db "select max(map_id) from Node"
```

hellovuong, 2023-04-25:
I figured it out that using sqlite lib directly is more suitable for my purpose. thank you for your help!

closed 2023-04-25 by hellovuong