introlab/rtabmap #1022: unabl to locate node rtabmap_ros

amoghv17, 2023-04-21:
The paths has been source correctly, and all files are set to excuteable. But while running the RTABMAP slam using ZED 2i camera I get this error.
[image]

OS: Ubuntu 20
ROS- noetic
ZED4.0sdk
I also have restarted the system several times and reinstalled the packages using the "sudo apt-get ros-noetic-rtabmap ros-notic-rtabmap-ros"
Thanks in advance

antoniskara, 2023-04-21:
Same problem here. Must be an issue with the latest noetic update

matlabbe, 2023-08-20:
Nodes have been changed of packages in latest version. See migration guide here: http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Migration_Guide_New_Interface_Noetic.2FROS2