introlab/rtabmap #1025: Issues Launching rtabmap_slam rtabmap on raspi 4 with ros

patrickH7561, 2023-04-24:
Hi All,

I am working on developing a mapping program for a lunar robotic project I am attempting to use rtabmaping on a raspi 4. I have been able to successful build my catkin workspace however immediately after the node launches I end up with a segfault. Really not sure what to do about this issue any help is appreciated.

[image]

matlabbe, 2023-04-24:
Do you have more than one OpenCV installed?

patrickH7561, 2023-04-24:
Definitely could have more than one Open-CV I'm a bit of a rookie here how would i check that?

xxEoD2242, 2023-04-25:
> Definitely could have more than one Open-CV I'm a bit of a rookie here how would i check that?

Throw it in in a Docker container and try it. The other way to check is /use/include and look for OpenCV