introlab/rtabmap #1029: coordinates of the trajectory of the camera_link

matuzawa1, 2023-04-30:
How can I get the coordinates of the trajectory of the camera_link?

matlabbe, 2023-04-30:
With `rtabmap-databaseViewer`, when doing "File->Export Poses...", you can then choose in which frame you want the poses. The `camera` frame should be there (default is base frame).

Genozen, 2023-05-01:
> With `rtabmap-databaseViewer`, when doing "File->Export Poses...", you can then choose in which frame you want the poses. The `camera` frame should be there (default is base frame).

@matlabbe , I had issue with the exported poses where they are no where near what was displayed on the Rtabmap status during a trial run.

reference: http://official-rtab-map-forum.206.s1.nabble.com/Extract-Odometry-Pose-from-Rtabmap-database-viewer-td9452.html