introlab/rtabmap #1036: Issue with squeeze image

borongyuan, 2023-05-23:
Hi, I'm trying to solve the FOV issue as mentioned in https://github.com/introlab/rtabmap/pull/1034#issuecomment-1557043379.
I've tested the usage of alphaScaling. The projection matrix can be calculated by cv::getOptimalNewCameraMatrix() correctly. But these data and images don't seem to work with rtabmap. I'm still looking for the reason, but the most likely reason is that using alphaScaling makes the image squeeze. That is to say, for the camera matrix calculated by cv::getOptimalNewCameraMatrix(), fx≠fy.
Here are the parameters I get.
left: fx=329.156006 fy=474.943451 cx=613.588454 cy=393.230591 baseline=0.075324
Here are the rectified left and right images. I put a coke in front of the camera, so you can see the squeeze.
[image]
[image]

matlabbe, 2023-05-30:
> But these data and images don't seem to work with rtabmap

What are you observing? You may compare left and right images with transparency overlay (e.g., using Gimp with two layers) to confirm if at least there is no vertical disparity. The squeeze effect would also modify the optical center of the calibration matrix.

To use `alphaScaling`, do you mean that depthai SDK gives only the original calibration, and we have to use `cv::getOptimalNewCameraMatrix()` externally to recover the matching calibration of the rectified images (using alpha onboard)? How can you make sure that the recovered matrices from getOptimalNewCameraMatrix are the same generated on the device?

cheers,
Mathieu

borongyuan, 2023-05-30:
> What are you observing?

It seems that somehow the points are not projected correctly. I've been dealing with other issues, and now I can look into this. I will create a separate branch for testing so you can take a look.

> To use `alphaScaling`, do you mean that depthai SDK gives only the original calibration, and we have to use `cv::getOptimalNewCameraMatrix()` externally to recover the matching calibration of the rectified images (using alpha onboard)?

In the [API documentation](https://docs.luxonis.com/projects/api/en/latest/references/cpp/#_CPPv2N3dai4node11StereoDepth15setAlphaScalingEf), there is only this paragraph. 

> Free scaling parameter between 0 (when all the pixels in the undistorted image are valid) and 1 (when all the source image pixels are retained in the undistorted image). On some high distortion lenses, and/or due to rectification (image rotated) invalid areas may appear even with alpha=0, in these cases alpha < 0.0 helps removing invalid areas. See getOptimalNewCameraMatrix from opencv for more details.

So I tested against these hints to verify that I understand correctly.

> How can you make sure that the recovered matrices from getOptimalNewCameraMatrix are the same generated on the device?

My tests are based on a modification of [Stereo Depth Video](https://github.com/luxonis/depthai-core/blob/main/examples/StereoDepth/stereo_depth_video.cpp). I set DepthAlign and alphaScaling, and get rectified images from the camera. At the same time I use getOptimalNewCameraMatrix to calculate parameters and apply these parameters to raw images using OpenCV. I get the same rectified images, indicating that this part of the calculation should be correct. I was most skeptical about the translation calculation of the right camera at first. Because this part of error will not be reflected on rectified images. But in the initialization, this part is calculated by the fx of the projection matrix and baseline, so it should be fine.

borongyuan, 2023-05-30:
I added alphaScaling-related code to [this branch](https://github.com/borongyuan/rtabmap/tree/alphascaling). At present, it is necessary to modify the value of alphaScaling in [L221](https://github.com/borongyuan/rtabmap/blob/aedfcdc57643799926e110033fd361355c335208/corelib/src/camera/CameraDepthAI.cpp#L221) and [L275](https://github.com/borongyuan/rtabmap/blob/aedfcdc57643799926e110033fd361355c335208/corelib/src/camera/CameraDepthAI.cpp#L275). We'll make it configurable if this issue is solved. I noticed that InliersRatio is extremely low with stereo input. But with RGB-D it works fine. This still implies that there are some problems with right camera params. Also weird is that it works when alpha is set to 1.0. I still need to find out the reason. After that, it may be better to filter out the black areas of the images, which are invalid.

borongyuan, 2023-05-31:
I think the problem has been located, [here](https://github.com/introlab/rtabmap/blob/bfc4e939d10976d0e7cd317f0257678a28ec87d3/corelib/src/util3d.cpp#L2521). The previous formula used by image_geometry/src/stereo_camera_model.cpp assumes fx=fy. Later [commit](https://github.com/ros-perception/vision_opencv/commit/52471a7141a9a38e3248134418c244d46dd28721) modified the implementation to support different fx and fy. I'll see how to do the same modification.

borongyuan, 2023-05-31:
After some derivation, I found that this is easy to solve with a slight modification.

matlabbe, 2023-06-05:
I did some comparisons with my OAK-D camera.

Camera matrix (`calibHandler.getCameraIntrinsics()`, without using `alpha` using old code):
```
[851.957763671875, 0, 639.60107421875;
 0, 852.2655639648438, 403.4123229980469;
 0, 0, 1]
```

New camera matrix (`getOptimalNewCameraMatrix()`, using `alpha` parameter with your [PR](https://github.com/introlab/rtabmap/pull/1042)):
 * `alpha=-1`
    ```
    [894.5590209960938, 0, 640.813128593767;
     0, 880.1116943359375, 403.1262598163412;
     0, 0, 1]
    ```

 * `alpha=0`
    ```
    [866.1574096679688, 0, 638.926311080897;
     0, 858.2002563476562, 403.1758010734538;
     0, 0, 1]
    ```
 * `alpha=0.5`
    ```
    [851.9566040039062, 0, 637.982902324462;
     0, 847.2445373535156, 403.2005717020102;
     0, 0, 1]
    ```

 * `alpha=1`
    ```
    [837.7557983398438, 0, 637.039493568027;
     0, 836.288818359375, 403.2253423305665;
     0, 0, 1]
    ```

It is strange that when `alpha=-1` (based on [opencv doc](https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6) referred by depthai), we don't get the same matrix without using alpha (old sdk version). When `getOptimalNewCameraMatrix()` should be used, only when `alpha` is used? Always?

**Is it because the `calibHandler.getCameraIntrinsics()` returns the pre-stereo rectified versions of the intrinsics**, not the new rectified camera intrinsics (after stereo rectification)? If so, the code was bugged before as I assumed that the intrinsics returned by that function were the stereo rectified intrinsics. It seems `getOptimalNewCameraMatrix()` should have been used before too. 

Note sure why in [depthai's API](https://docs.luxonis.com/projects/api/en/latest/references/cpp/#_CPPv2N3dai4node11StereoDepth15setAlphaScalingEf) `alpha=0` can still have black borders, while [OpenCV's API](https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6) `alpha=0` is no black border. I remember doing some tests with depthai calibration tool, and using opencv rectification with alpha=0, there were no black borders: https://github.com/luxonis/depthai/issues/853

I'll try later reproducing https://github.com/luxonis/depthai/issues/853 and compare the intrinsics during calibration process and the ones returned by `calibHandler.getCameraIntrinsics()` to confirm if `getOptimalNewCameraMatrix()` shoudl always be called independently of the alpha parameter used or not.

borongyuan, 2023-06-05:
> It is strange that when `alpha=-1` (based on [opencv doc](https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6) referred by depthai), we don't get the same [...]

My recent commits are based on depthai-core release v2.21.2. Some features such as alphaScaling and set CameraExposureOffset for timestamp may not be supported in previous versions. Behavior may also change in future releases. Yes, for current depthai version, its behavior of `alpha=-1` is different from OpenCV. But when alpha is in [0,1], their behavior is consistent. `getOptimalNewCameraMatrix()` should be used, only when `alpha` is used. We can indeed add an option to decide whether to use `alpha`, or handle it separately when `alpha=-1`. But I think this is a bit redundant, because the FOV increase of using `alpha` is obvious for OAK cameras. This may reduce some long-distance measurement accuracy, but for VSLAM, we prefer large FOV. So my suggestion is to always use `alpha≥0`, and therefore always use `getOptimalNewCameraMatrix()`.

> **Is it because the `calibHandler.getCameraIntrinsics()` returns the pre-stereo rectified versions of the intrinsics**, not the new rectified camera intrinsics (after stereo rectification)? If so, [...]

In order to express the behavior of depthai clearly, we refer to the definition of [CameraInfo](https://wiki.ros.org/image_pipeline/CameraInfo) in ROS. The `calibHandler.getCameraIntrinsics()` actually returns **K** camera matrix. But depthai does not use `cv::stereoRectify()` to calculate **P** projection matrix. Instead it directly uses **K** of aligned camera as **P** (with different translation offset) for all images. That is, if depth is set to align to `DepthAlign::RECTIFIED_LEFT`, then **K** of left camera is used as **P** for rectified_left, rectified_right and depth. If `alpha` is used, then new_camera_matrix of left camera is used as **P**.

> Note sure why in [depthai's API](https://docs.luxonis.com/projects/api/en/latest/references/cpp/#_CPPv2N3dai4node11StereoDepth15setAlphaScalingEf) `alpha=0` can still have black borders, while [OpenCV's API](https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6) `alpha=0` is no black border. I remember doing some tests with depthai calibration tool, and using opencv rectification with alpha=0, there were no black borders: [luxonis/depthai#853](https://github.com/luxonis/depthai/issues/853)

Since new_camera_matrix is calculated for the left camera, when `alpha=0`, the valid area of the left camera will be maximized. Black borders will appear on rectified_right, as you can see from my first post. In fact, there is a little black border in the middle of the bottom of rectified_left from my first post. Probably only a pixel or two, barely noticeable. I don't know if this is what their documentation refers to.

Here are some tests about alphaScaling.
1. `alpha` not set (disabled)
[image]
2. `alpha=0`, perception area is much larger
[image]
3. `alpha=1`, perception area is further enlarged, a lot of black borders
[image]
I think `alpha=0` is a good choice. When `alpha=1`, the perception area does not increase much more, but the invalid area is much more. And the extra areas don't provide too many features. The edges of black borders do add invalid features.

matlabbe, 2023-06-05:
Thanks for the explanation, I agree the default alpha value should be 0. Not sure why depthai doesn't use [stereoRectify](https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6) function directly, that would make sure that both left and right images don't have invalid pixels (because they can cause wrong disparity). I'll finish reviewing the PR tonight and merge it, very nice stuff!

PR #1042 (2023-06-06): Support cameras with different fx and fy, enable alphaScaling for OAK cameras
Fix #1036 
This makes better use of the FOV of wide-angle lens.
closed 2023-06-06 by matlabbe