introlab/rtabmap #1041: Filtering Loop Closures and Limiting Corrections in ROS Application

rmunozor2703, 2023-05-31:
Hello everyone!

I wanted to express my appreciation for the incredible work done on this framework. It's truly amazing how user-friendly and adaptable it is to meet your specific needs. Congrats to the developers!

Now, I have a question regarding an application I'm working on that involves using ROS. I'm wondering if there's a way to filter out loop closures that fall outside of a specific area and only consider the ones that are within that defined region. This functionality would greatly enhance the accuracy and reliability of my application.

Additionally, I'm curious if it's possible to impose limits on the corrections made by the system. For instance, if the detected loop closure would result in a camera position adjustment of just a few centimeters, I would prefer not to apply those corrections. This precaution is crucial to maintain a seamless and immersive experience within the application.

I would greatly appreciate any guidance or suggestions on how to achieve these goals. Thank you all in advance for your valuable input!

matlabbe, 2023-06-04:
> I'm wondering if there's a way to filter out loop closures that fall outside of a specific area and only consider the ones that are within that defined region.

Can you give an example? Do you mean disabling loop closure detection if a robot or a person (VR/scanning) is exiting a fixed perimeter?

> if the detected loop closure would result in a camera position adjustment of just a few centimeters, I would prefer not to apply those corrections.

rtabmap has loop closure validation check in case the loop closure adds too much deformation in the graph (which could mean is a bad loop closure). In you case it seems the inverse, do you would want to ignore loop closures if they don't deform a lot the graph? I think you may take a look at [REP105](https://www.ros.org/reps/rep-0105.html), about the purpose of the map and odom frame. I think in your application you seem to care about more the odom frame than the map frame. For example, to avoid localization jumps, a robot controller would use the odom frame instead of the map frame, and eventually update its goal at lower frequency based on map frame. The control loop would be done in the smooth odom frame. For your case, it seems that you could estimate the trajectory in odom frame, then use map corrections only when you detect that correction map->odom changed more than a few cm.

matlabbe, 2023-06-11:
To disable loop closure detection, you could do:
```
rosparam set /rtabmap/rtabmap/Kp/MaxFeatures -1
rosservice call /rtabmap/update_parameters
```
To re-enable loop closure detection:
```
rosparam set /rtabmap/rtabmap/Kp/MaxFeatures 500
rosservice call /rtabmap/update_parameters
```

I think for VR, some approaches may smooth the localization jumps over a couple of seconds so that the user won't notice that the model is moved back to right position ([similar trick used here to manipulate the same block](https://www.youtube.com/watch?v=v-5u0z4zA_8), I guess for localization maybe something similar could be done). In ROS, that smoothed transform could be added above `map` frame (`map_smooth->map->odom->base_link...`), or you replace `map` frame (`publish_tf=false` for `rtabmap` node) by yours smoothed.

 For MR/AR/XR (augmented reality content), if a simulated wall is slowly moving to the real wall for a couple of seconds, it may be noticeable. Maybe when the person is not looking at that wall, you jump it back at the right position. 

cheers,
Mathieu

closed 2023-07-04 by rmunozor2703