introlab/rtabmap #1043: unable to build rtabmap with python3 support

GVMafeedme, 2023-06-01:
I am currently building a docker image to run rtabmap_ros with Superpoint. The problem is that it needs to be built with  -DWITH_PYTHON=ON, and when I do so, the following error appears:
```
#64 211.2 [ 98%] Linking CXX executable ../../bin/rtabmap-odometryViewer
#64 211.2 In file included from /rtabmap/examples/WifiMapping/main.cpp:40:
#64 211.2 /rtabmap/corelib/src/../include/rtabmap/core/PythonInterface.h:14:10: fatal error: Python.h: No such file or directory
#64 211.2    14 | #include <Python.h>
#64 211.2       |          ^~~~~~~~~~
#64 211.2 compilation terminated.
#64 211.2 make[2]: *** [examples/WifiMapping/CMakeFiles/wifi_mapping.dir/build.make:71: examples/WifiMapping/CMakeFiles/wifi_mapping.dir/main.cpp.o] Error 1
#64 211.2 make[1]: *** [CMakeFiles/Makefile2:1732: examples/WifiMapping/CMakeFiles/wifi_mapping.dir/all] Error 2
#64 211.2 make[1]: *** Waiting for unfinished jobs....
#64 211.2 In file included from /rtabmap/examples/RGBDMapping/main.cpp:44:
#64 211.2 /rtabmap/corelib/src/../include/rtabmap/core/PythonInterface.h:14:10: fatal error: Python.h: No such file or directory
#64 211.2    14 | #include <Python.h>
#64 211.2       |          ^~~~~~~~~~
#64 211.2 compilation terminated.
```

My objective is to run Superpoint with Superglue, but the Python interface does not seem to find Python, even though I have ```python3-dev``` installed. 
I have tried using ```python3.8-dev``` and ```python-dev``` but nothing works.
Other information:
- Environment: docker nvidia/cuda:11.3.1-cudnn8-devel-ubuntu20.04
- ROS: Noetic
- OpenCV: 4.5.2
- Rtabmap version: 0.21.1

closed 2023-06-01 by GVMafeedme

mattiasmar, 2023-06-11:
@GVMafeedme How did you solve this problem? Could you share you working dockerfile?

matlabbe, 2023-06-11:
@mattiasmar there is the one I tested some time ago: https://github.com/introlab/rtabmap/issues/896#issuecomment-1257085425 (I don't have a final docker file for that, but the manual steps may be done in a new Dockerfile to automate the process)